slic3r/Slic3r #4635: perl-local missing from tar.bz2

mrvanes, 2018-12-04:
Unable to start Slic3r after downloading 2018-12-3 version of Slic3r tar.bz2 (latest).

Slic3r bashfile refers to $DIR/perl-local which doesn't exist in the tar? If I copy perl-local from previous downloaded version, perl can't locate strict.pm and fails to start.

G-Pereira, 2018-12-05:
I'm having the same issue on Arch Linux installing the package slic3r-dev

lordofhyphens, 2018-12-05:
Sound refactored the build+deployment on Travis and it broke something that
wasn't noticed immediately. When I have time again I will look into it.

lordofhyphens, 2018-12-30:
@G-Pereira We don't maintain the arch package, so I can't help you at all.

msrd0, 2018-12-30:
@lordofhyphens @G-Pereira The Arch package does nothing more than downloading the latest available tar-gz-archive for the master branch from dl.slic3r.org, extract it and link the `slic3r` executable to `/usr/bin`. If this issue was resolved, and the `perl-local` executable was in the archive again, the ArchLinux package would be working again.

lordofhyphens, 2018-12-31:
This should be sorted out now. @msrd0 I'd appreciate it if more people could test the new appimages and tarballs.
https://dl.slic3r.org/dev/linux/branches/Slic3r-1.3.1-dev-8531859-PR4659-x86_64.AppImage
https://dl.slic3r.org/dev/linux/branches/slic3r-1.3.1-dev-8531859-linux-x64-PR4659.tar.bz2

WorksForMe(tm) on my Debian sid system.

PR #4659 (2018-12-31): Update packaging script to be more intelligent
The packaging script isn't finding perl (so it can't include it). 

Fixes #4635
closed 2018-12-31 by lordofhyphens