slic3r/Slic3r #4636: Invalid estimated time display before/during print

VanessaE, 2018-12-05:
**Describe the bug**

Before and during a print, there's no valid time displayed.  Instead, it shows "NaN hours and NaN minutes".

**To Reproduce**
Slice and print something :smiley:

**3D Model and Slic3r Configuration Export**
Any model, any config (but, [config-20181203.ini.zip](https://github.com/slic3r/Slic3r/files/2646712/config-20181203.ini.zip) if you really want one).

**Expected behavior**
There should be an estimated time displayed instead of "NaN hours/minutes".

**Screenshots**
[image]

**Desktop (please complete the following information):**
XFCE 4.12, Debian "Buster", Slic3r commit 21eb603cc16946b14e77d3c10cbee2f1163503c6

ted1735, 2019-02-21:
Same issue here on stable version as well as latest dev version

VanessaE, 2019-08-17:
No opinion on this issue?