slic3r/Slic3r #4637: Can't Pickup Correct Extruder in Start G-code

UticaTechClub, 2018-12-05:
**Describe the bug**
**Start G-code** custom section runs before `[current_extruder]` is assigned.
I have a four-extruder printer with extruders docked on the back wall. 
Before printing can start, cart needs to go to the first used extruder and take it.

**To Reproduce**
Slice any job.

**3D Model and Slic3r Configuration Export**
Here is my **Start G-code** with two extruders under the test:
```
G28 ;Home
G29 ;Auto Level
G92 Z6.0 ;At this point Z should be 10mm, which is too far. Declare it to be thought as 6mm. Greater value makes nozzle closer to the bed.
M218 T0 X0.0 Y0.0 Z0.0 ;Set first extruder with no offset.
M218 T1 X0.0 Y0.0 Z0.4 ;Set second extruder offset. Positive Z makes bed closer to nozzel.

;Take current tool---
G1 Y220 F4800
{if [current_extruder] == 0}G1 X1.6 F4800 ;Extruder 0 location
{if [current_extruder] == 1}G1 X83 F4800 ;Extruder 1 location
G1 Y315 F4800
M400
M280 P1 S130 ;Lock servo
M0 P1000 ;Wait untill servo turns
G1 Y220 F4800
;---Take current tool
```

**Expected behavior**
[current_extruder] should be initialized before Start G-code is called.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: [Windows]
 - Version [e.g. 1.3.0]

**Additional context**
When extruders are switching, the logic works perfectly based on `[previous_extruder]` and `[next_extruder]` variables.
When printing is done, the logic also perfectly works, because I can dock the last used extruder in **End G-code** section based on `[current_extruder]` variable.
Problem is only at the beginning of the print.

supermerill, 2018-12-06:
Hello

> Before printing can start, cart needs to go to the first used extruder and take it.

The problem is that you can't add lines in the start gcode section to select the right tool for homing?

UticaTechClub, 2018-12-06:
@supermerill 
Hi,

> The problem is that you can't add lines in the start gcode section to select the right tool for homing?

No, homing and bed leveling is done without the tool as end-stops are mounted on the frame and the bed sensor is on the cart itself, so that is not a concern.

Right tool/extruder needs to be picked-up for printing and I don't know which tool/extruder to pick because `[current_extruder]` is not initialized yet. Pick-up location is tool/extruder specific. For example:
```
{if [current_extruder] == 0}G1 X1.6 F4800 ;Extruder 0 location
{if [current_extruder] == 1}G1 X83 F4800 ;Extruder 1 location
{if [current_extruder] == 2}G1 X181 F4800 ;Extruder 2 location
{if [current_extruder] == 3}G1 X262 F4800 ;Extruder 3 location
```
To generalize:
Starting gcode needs to know which tool/extruder is the first one (or the initial one) for a printing job so it can go and take it.

UticaTechClub, 2018-12-06:
Just to clarify...

**(Doesn't work!!!)** Start code (use ??? to pickup the first tool)

**(Works)** Tool change 1 code (use `[previous_extruder]/[next_extruder]` to park and pickup the tool)
**(Works)** Tool change 2 code (use `[previous_extruder]/[next_extruder]` to park and pickup the tool)
...
**(Works)** Tool change N code (use `[previous_extruder]/[next_extruder]` to park and pickup the tool)

**(Works)** End code (use `[current_extruder]` to park the last tool)

lordofhyphens, 2018-12-06:
@supermerill the issue is that the current extruder is not exposed during start gcode (and I think it can and should be).

supermerill, 2018-12-06:
ok.
It's an e3D tool-changer?

UticaTechClub, 2018-12-06:
> ok.
> It's an e3D tool-changer?

We have been working with student on ours for a long time now, well before e3d did theirs, but our system is very similar to e3d's from this prospective and e3d must be facing the same issue.

UticaTechClub, 2019-01-09:
I have developed a fix for this issue.
I tested it and now trying to make a pull request.

I am somewhat new to GitHub, and encountered a problem pushing into Slic3r repo. 
Please help me if you can?

`
$ git clone https://github.com/slic3r/Slic3r.git
`

got a fresh clone...

`
$ git checkout -b issue-4637
`

ok, made my branch and did edits...

`$ git status`
`On branch issue-4637`
`Changes not staged for commit:`
`  (use "git add <file>..." to update what will be committed)`
`  (use "git checkout -- <file>..." to discard changes in working directory)`
`  modified:   lib/Slic3r/Print/GCode.pm`
`  modified:   xs/src/libslic3r/PrintGCode.cpp`
`no changes added to commit (use "git add" and/or "git commit -a")`

`$ git add lib/Slic3r/Print/GCode.pm`
`$ git add xs/src/libslic3r/PrintGCode.cpp`
`$ git commit -m "fix to make current_extruder available in start GCode issue 4637"`

and did the commit
ok, all good and now lets push...

`$ git push --set-upstream origin issue-4637`
`remote: Permission to slic3r/Slic3r.git denied to UticaTechClub.`
`fatal: unable to access 'https://github.com/slic3r/Slic3r.git/': The requested URL returned error: 403`

It will not push. I tried everything I could.
I even tried making my own GitHib project and was able to successfully push into it.

Do you guys have any hints for me???

lordofhyphens, 2019-01-09:
Open the pull request from your fork to us.
https://help.github.com/desktop/guides/contributing-to-projects/creating-a-pull-request/

UticaTechClub, 2019-01-09:
> Open the pull request from your fork to us.
> https://help.github.com/desktop/guides/contributing-to-projects/creating-a-pull-request/

Yes, I understand, but I can't push my branch and thus can't continue into creating pull request:

`$ git push --set-upstream origin issue-4637`
`remote: Permission to slic3r/Slic3r.git denied to UticaTechClub.`
`fatal: unable to access 'https://github.com/slic3r/Slic3r.git/': The requested URL returned error: 403`

UticaTechClub, 2019-01-09:
I could even attached modified source code files here, but I want to do it the right way.

You guys are greatly appreciated for an amazing work that is done here.

Marlin is another comparable effort making us 3D printer developers very thankful.

You guys should consider coming to RepRap Midwest conferance this year:
[https://www.eventbrite.com/e/2019-midwest-reprap-festival-mrrf2019-tickets-51075814135](url)

lordofhyphens, 2019-01-10:
Sorry, after you said that you made your own fork and could push to it I thought that referencing the GitHub docs was enough (as I thought you had done that important prerequisite). 

The normal way to do things is to add your GitHub fork as a remote (`git remote add https://github.com/username/Slic3r myfork`) and push to it. 

Then create the pull request from the branch on your fork to master on slic3r/Slic3r.

nwneisen, 2019-09-04:
I came across this issue while looking at #4338.  Went ahead and create a PR to take care of it.