slic3r/Slic3r #4638: Speed issue with Slic3r

athulsnair, 2018-12-05:
Hello,

I am trying to print with clay.
These are my Slic3r speed settings,
[image]

Other information;
Nozzle: 1.4 mm
First layer height: 0.9 mm
layer height: 1 mm
Filament dia: 1.8 mm
perimeter: 3

but when I print, First layer is printed  faster than remaining layers. And in remaining layers, two inner perimeters are printed much slower than outer perimeter.

Video: [https://drive.google.com/open?id=1UE2MFhjTXpVEirsNK3KlALi4Fb5Ca8rX](url)
Gcode: [https://drive.google.com/open?id=1GMyNPaX8YiB3-P2UNHGC3gTZC3M3oilm](url)

Isn't this strange since i give print speed same for all.

My slic3r version is: 1.3.1 dev

athulsnair, 2018-12-06:
I thought it might be a firmware problem too. It's not.

The problem is with the "Filament Settings" of Slic3r

[new_config-20181206T070832Z-001.zip](https://github.com/slic3r/Slic3r/files/2652033/new_config-20181206T070832Z-001.zip)

This is a new configuration. Use "Filament settings" from this config. It is same as initially uploaded configurations "Filament Settings", but this one prints at proper speed

foreachthing, 2018-12-06:
@athulsnair I noticed that, too!
But now ... what was the problem exactly?

VanessaE, 2018-12-06:
Please attach an example model also.

You have auto-cooling enabled with a minimum layer time of 30 seconds. If a layer should take less than that, Slic3r will slow down internal perimters, infill, etc. to kill time, leaving exterior perimeters at their configured speed.  Your video suggests exactly that.

athulsnair, 2018-12-07:
@VanessaE Model can be found inside the zip file that I attached to first comment. May be it has to do with the cooling. I will check that.

@foreachthing I though it might be a bug. At higher nozzle size and layer height, it fails to follow the print speed given.
I make "cooling" settings same for both and reply

VanessaE, 2018-12-07:
Oh right, not sure how I missed that model.  Loaded and sliced with one of my configs, and as expected, small layer times.  With two perimeters, zero infill, no auto-cooling/slow-down, and a flat 30 mm/s, I get layer times in the 8 second range.

closed 2018-12-31 by VanessaE