slic3r/Slic3r #4640: External perimeters not given adequate detail

VanessaE, 2018-12-09:
**Describe the bug**

This is Slic3r commit 21eb603cc16946b14e77d3c10cbee2f1163503c6 plus @supermerill's thin wall improvements PR (I'm at commit ca9f3f7e29662c25a0bd5c3ed00232ea8ed68d2a of that).

Slic3r's discarding small features in the model instead of trying to recreate them.  Consider the nameplate on the back of a Benchy:

[image]

0.4 mm nozzle, 0.4 mm line width, 0.08 mm layer height (config: [config-20181209.ini.zip](https://github.com/slic3r/Slic3r/files/2660290/config-20181209.ini.zip) ).

Compare to this photo I found on Hackaday.  While the image and print quality are mediocre, it clearly shows all of the lettering's individual strokes.  This was apparently also printed with a 0.4 mm nozzle, sliced with Cura (not sure about the layer height, but it looks to be about 0.1 mm).

[image]

(rotated, cropped, resized, and I painted-out a blob that was in the middle of the plate, for easier visual comparison; source:  https://hackaday.com/2016/08/30/review-monoprice-maker-ultimate-3d-printer/ ).

supermerill, 2018-12-10:
for the  '#' : the thin area between the two '/' is too small for a 0.4 nozzle to do a U.

But there should be an extrusion between 3,D,b

If you disable the thin_wall option, you can show it.

The "culprit" is done at perimetergenerator.cpp l85 vs l97. I didn't change it (i just added a little post-process for thin wall to relax the algo a bit if it creates too much ares).

What  i understand : If no thin wall, it lay every perimeter the best way possible. If thin wall, it exclude areas that can be too thin for 2 perimeters (you can se it, when 2 perimeter are almost one over another -> overextrusion), and pray for the thin walls to extrude these thin area. The problem is: sometimes, these areas are too tiny.

I will try something.

VanessaE, 2018-12-10:
This is one of those situations where Cura, for example, would probably do those overlapped perimeters I mentioned.

VanessaE, 2018-12-21:
Looking much better with the latest updates on the thin walls PR (commit 44e7ec855e1706fe1af6dddaff5835e018d0b69d):

[image]

closed 2019-07-30 by VanessaE