slic3r/Slic3r #4643: Segfault libX11

platsch, 2018-12-11:
**Describe the bug**
I*m currently trying to build Slic3r (master: 21eb603cc16946b14e77d3c10cbee2f1163503c6) on a fresh Ubuntu 18.04 (4.15.0-42-generic). All prerequisites installed, Build.PL completes without errors. I can run Slic3r from terminal, but when starting with gui it just returns a segfault.

Tried running it in gdb (output below). Seems to be a problem with wx/GL, maybe Wayland related?
Did anyone successfully build on Ubuntu 18.04? Any ideas on how to further debug? I'm clueless at the moment...

```
(gdb) run slic3r.pl
Starting program: /usr/bin/perl slic3r.pl

Thread 1 "perl" received signal SIGSEGV, Segmentation fault.
0x00007fffefdf46b9 in XQueryExtension () from /usr/lib/x86_64-linux-gnu/libX11.so.6
(gdb) bt
#0  0x00007fffefdf46b9 in XQueryExtension () at /usr/lib/x86_64-linux-gnu/libX11.so.6
#1  0x00007fffd7dd8252 in  () at /usr/lib/x86_64-linux-gnu/libGLX.so.0
#2  0x00007fffd7dd42d5 in glXQueryVersion () at /usr/lib/x86_64-linux-gnu/libGLX.so.0
#3  0x00007fffd7910ac5 in wxGLCanvasX11::GetGLXVersion() () at /usr/lib/x86_64-linux-gnu/libwx_gtk3u_gl-3.0.so.0
#4  0x00007fffd7911165 in wxGLCanvasX11::ConvertWXAttrsToGL(int const*, int*, unsigned long) () at /usr/lib/x86_64-linux-gnu/libwx_gtk3u_gl-3.0.so.0
#5  0x00007fffd7911998 in wxGLCanvasX11::InitXVisualInfo(int const*, __GLXFBConfigRec***, XVisualInfo**) () at /usr/lib/x86_64-linux-gnu/libwx_gtk3u_gl-3.0.so.0
#6  0x00007fffd7912aa0 in wxGLCanvas::Create(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&, int const*, wxPalette const&) () at /usr/lib/x86_64-linux-gnu/libwx_gtk3u_gl-3.0.so.0
#7  0x00007fffd7912c43 in wxGLCanvas::wxGLCanvas(wxWindow*, int, int const*, wxPoint const&, wxSize const&, long, wxString const&, wxPalette const&) () at /usr/lib/x86_64-linux-gnu/libwx_gtk3u_gl-3.0.so.0
#8  0x00007fffe400420f in  () at /usr/lib/x86_64-linux-gnu/perl5/5.26/auto/Wx/GLCanvas/GLCanvas.so
#9  0x0000555555630441 in Perl_pp_entersub (my_perl=0x555555954260) at pp_hot.c:4231
#10 0x0000555555628266 in Perl_runops_standard (my_perl=0x555555954260) at run.c:41
#11 0x00005555555a118f in Perl_call_sv (my_perl=0x555555954260, sv=0x555559063300, flags=<optimized out>) at perl.c:2831
#12 0x00007fffe40035ba in  () at /usr/lib/x86_64-linux-gnu/perl5/5.26/auto/Wx/GLCanvas/GLCanvas.so
#13 0x0000555555630441 in Perl_pp_entersub (my_perl=0x555555954260) at pp_hot.c:4231
#14 0x0000555555628266 in Perl_runops_standard (my_perl=0x555555954260) at run.c:41
#15 0x00005555555a136f in Perl_call_sv (my_perl=<optimized out>, sv=0x5555587b5088, flags=<optimized out>) at perl.c:2848
#16 0x00007ffff2b12118 in  () at /usr/lib/x86_64-linux-gnu/perl5/5.26/auto/Wx/Wx.so
[...]
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install Ubuntu 18.04 864
2. Follow https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux to build Slic3r from github
3. run perl slic3r.pl
4. Segmentation fault

**3D Model and Slic3r Configuration Export**
Not required

**Expected behavior**
Slicer should launch and show the GUI

**Desktop (please complete the following information):**
- Ubuntu 18.04 4.15.0-42-generic
- Github master 21eb603cc16946b14e77d3c10cbee2f1163503c6

lordofhyphens, 2018-12-11:
Sigh, now I have to spin up an Ubuntu vm.

platsch, 2018-12-11:
Or you just hope for me digging further into this and come up with a solution :-)
Any clue where to start?

supermerill, 2018-12-11:
No problems on my 16.04.

lordofhyphens, 2018-12-11:
You could try to build cppgui with -DEnable_GUI, which does instantiate OpenGL.

platsch, 2018-12-12:
Seems to be something hardware and perl related.
I don't get the issue on a different (hardware) system.
Also, building with perlbrew threaded-perl-5.26.1 works on my laptop, I just had to install a missing
`libssl-dev` package.

platsch, 2018-12-13:
I consider this edge-case as not crucial and close the issue as solved. At least for me it isn't worth further investigation.

closed 2018-12-13 by platsch