slic3r/Slic3r #4645: Not running on linux with rx580

PiPerson, 2018-12-14:
Hi
I am running Ubuntu 18.04, and when I try to run the slic3r appimage, I get these errors:
`person@person-pc:~$ /home/person/Downloads/Slic3r-1.3.0-x86_64.AppImage

(slic3r.pl:7692): Gtk-WARNING **: 18:53:31.741: Unable to locate theme engine in module_path: "adwaita",

(slic3r.pl:7692): Gtk-WARNING **: 18:53:31.744: Unable to locate theme engine in module_path: "adwaita",
libGL error: unable to load driver: radeonsi_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: radeonsi
libGL error: unable to load driver: radeonsi_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: radeonsi
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
/tmp/.mount_Slic3rub8cQZ/AppRun: line 11:  7692 Segmentation fault      (core dumped) LD_LIBRARY_PATH="$DIR/usr/lib" "${DIR}/usr/bin/perl-local" -I"${DIR}/usr/lib/local-lib/lib/perl5" "${DIR}/usr/bin/slic3r.pl" --gui "$@"`

I have a rx580 graphics card with all the proper drivers.
I looked it up and found that peole suggested I preload the libstdc, so I ran
`person@person-pc:~$ LD_PRELOAD='/usr/lib/x86_64-linux-gnu/libstdc++.so.6' /home/benjymon/Downloads/Slic3r-1.3.0-x86_64.AppImage

(slic3r.pl:7834): Gtk-WARNING **: 18:56:31.610: Unable to locate theme engine in module_path: "adwaita",

(slic3r.pl:7834): Gtk-WARNING **: 18:56:31.613: Unable to locate theme engine in module_path: "adwaita",
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: radeonsi
mesa: for the -simplifycfg-sink-common option: may only occur zero or one times!
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: radeonsi
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
/tmp/.mount_Slic3rE5Cfae/AppRun: line 11:  7834 Segmentation fault      (core dumped) LD_LIBRARY_PATH="$DIR/usr/lib" "${DIR}/usr/bin/perl-local" -I"${DIR}/usr/lib/local-lib/lib/perl5" "${DIR}/usr/bin/slic3r.pl" --gui "$@"`

from this it would seem that the error is libgl related.

Thanks

lordofhyphens, 2018-12-31:
@PiPerson can you give the new AppImage a try? https://dl.slic3r.org/dev/linux

lordofhyphens, 2019-01-01:
@PiPerson https://dl.slic3r.org/dev/linux/branches/Slic3r-1.3.1-dev-4ff81a6-PR4666-x86_64.AppImage may work for you.

lordofhyphens, 2019-01-01:
It's probably related to: https://askubuntu.com/questions/654566/steam-doesnt-start-libgl-error

PiPerson, 2019-01-02:
Sorry for taking a while to respond.
I tried the new appimage and got the same error.
`/tmp/.mount_Slic3rqvlJkK/AppRun: line 41: : No such file or directory

(slic3r.pl:16121): Gtk-WARNING **: 20:03:25.062: Unable to locate theme engine in module_path: "adwaita",

(slic3r.pl:16121): Gtk-WARNING **: 20:03:25.066: Unable to locate theme engine in module_path: "adwaita",
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: radeonsi
mesa: for the -simplifycfg-sink-common option: may only occur zero or one times!
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: radeonsi
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
/tmp/.mount_Slic3rqvlJkK/AppRun: line 59: 16121 Segmentation fault      (core dumped) LD_LIBRARY_PATH="$DIR/usr/lib:${LD_LIBRARY_PATH}" "${DIR}/usr/bin/perl-local" -I"${DIR}/usr/lib/local-lib/lib/perl5" "${DIR}/usr/bin/slic3r.pl" --gui "$@"`

lordofhyphens, 2019-01-03:
Any ideas @probonopd  ?

probonopd, 2019-01-03:
Same as #4663?

probonopd, 2019-01-03:
Please try:

```
./Slic3r-1.3.1-dev-4ff81a6-PR4666-x86_64.AppImage --appimage-extract 
 rm squashfs-root/usr/lib/swrast_dri.so 
 ./squashfs/AppRun
```

Does it run?

lordofhyphens, 2019-01-03:
I believe that I moved the file and tried to pull the same trick as libstdc++

https://github.com/slic3r/Slic3r/blob/master/package/linux/appimage-apprun.sh

This is in current master now.

lordofhyphens, 2019-01-03:
https://dl.slic3r.org/dev/linux/Slic3r-1.3.1-dev-39b157b-x86_64.AppImage

PiPerson, 2019-01-03:
I tried running the new dev appimage and sadly it still does not run
Also thanks for your help

prof7bit, 2019-01-04:
A workaround is mentioned here:

https://superuser.com/questions/1377555/slic3r-appimage-fails-to-run-with-libgl-error-unable-to-load-driver-radeonsi/1377556

It seems the bundled standard library is out of date, preloading it will make it run. 

BTW the Prusa edition appimages do NOT have this error, maybe it is worth having a look at how they are bundling their appimages.

lordofhyphens, 2019-01-06:
@prof7bit I don't bundle the standard libraries in a way in development builds that they would get used on most systems (using the Appimage-checkrt script). 

@PiPerson  
Try: 
download https://dl.slic3r.org/dev/linux/Slic3r-1.3.1-dev-39b157b-x86_64.AppImage
````
./Slic3r-1.3.1-dev-39b157b-x86_64.AppImage --appimage-extract 
 rm -rf squashfs-root/usr/optional
 ./squashfs-root/AppRun
````

lordofhyphens, 2019-01-06:
@PiPerson So those commands removed all external libraries that are usually system libraries. 

Does glxgears work (if it is installed)? 

https://www.linuxquestions.org/questions/showthread.php?p=5895937#post5895937

lordofhyphens, 2019-01-06:
@PiPerson Does the .tar.bz2 run for you?

PiPerson, 2019-01-08:
glxgears does run
But the master-latest .tar.bz2 does not run

lordofhyphens, 2019-01-08:
@PiPerson Interesting. I, at this point, have no idea what is going on.
Does this run?
````
LD_PRELOAD='/usr/lib/x86_64-linux-gnu/libstdc++.so.6' ./Slic3r-1.3.1-dev-f18c340-x86_64.AppImage
````

lordofhyphens, 2019-01-24:
Ooookay. I have no idea what changed, but glad to hear it works for you.

closed 2019-01-24 by lordofhyphens