slic3r/Slic3r #4647: bridging problem

ruggb, 2018-12-18:
OS: Windows 10 1803
Slic3r-master.2018.12.02.2123.21eb603c.64bit
it also occurs with this version Slic3r-master.2018.07.12.1773.5dbfc10.64bit
Repetier 2.1.3
Marlin 2.0 beta
Print file created in Sketchup 2017 make 

slic3r is set for 3 perimeters min == up to layer 200 it only makes 2 perimeters.
at layer 200 it creates another inside perimeter then fails to connect the bridge layer to the opposite side.
At some point the infill finally connects and it is ok past that point.
There is a taper on this part of the model and it almost appears like it is using the model size near the top for the bridge dimensions.
The print preview in Repetier shows it properly.
It then bridges again properly @ layer 261 and continues with 100% infill (Slic3r's choice).
I changed the infill overlap from 55% to 75% and it almost recovered at the 3rd layer of the bottom instead of part way thru the honeycomb infill.

Why is it only printing 2 perimeters instead of 3 at the bottom?
Why does it start to make the 3rd perimeter on the inside in mid-air at layer 200? (without much success)
Why is it not connecting the infill to the other side?

This problem is specific to this model.
[Legs.zip](https://github.com/slic3r/Slic3r/files/2692324/Legs.zip)

ruggb, 2019-01-03:
anyone?

supermerill, 2019-01-03:
> Why is it only printing 2 perimeters instead of 3 at the bottom?

it's only printing 1 perimeter. Because there are no "inside", the piece is very thin at this height.

> Why does it start to make the 3rd perimeter on the inside in mid-air at layer 200? (without much success)

because the piece is now very big and so it has to create the bottom surface.

> Why is it not connecting the infill to the other side?

because yous elect 3 perimeter and the slicer isn't super smart. Create a modifier cube, place it at this height and change the "perimeters" setting to 1 or 2 for it. See the slic3r doc for a more in-depth tutorial.

VanessaE, 2019-01-03:
This looks like a duplicate of https://github.com/slic3r/Slic3r/issues/3916

ruggb, 2019-01-03:
Thanks
that is the issue. Why can't the slicer figure out I am a dummy and asked for too many perimeters??

VanessaE, 2019-01-03:
Beats me.  As I stated in that linked issue, Slicer should just automatically reduce the number of bridging perimeters when it needs to.

closed 2019-01-03 by VanessaE

lordofhyphens, 2019-01-04:
I think that the "minimum shell thickness needs to be faster and/or more
reliable so it is more useful. I have noticed that one of the two does not
play nice with adaptive slicing (although I could have been misremembering
and it misbehaves more often).

supermerill, 2019-01-04:
> Why can't the slicer figure out I am a dummy and asked for too many perimeters??

Because nobody has coded the feature (yet)