slic3r/Slic3r #4651: Running perl Build.PL in a path containig umlauts results into errors

Taki-B, 2018-12-24:
When the cloned code resides in a directory path including umlauts (/home/taki/Geräte/3D-Drucker/Slic3r_GIT/Slic3r/ in my case) parts of the code won't compile.

Steps to reproduce the behavior:
1. Try to build Slic3r somewhere in a path with umlauts

**Desktop (please complete the following information):**
 - OS: Linux
 - Version 1.3.1-dev (`git describe --tag` 1.3.0-1243-g21eb603c)

**Additional context**
The bug is easy to circumvent: just rename the directory and everything runs as expected

lordofhyphens, 2018-12-31:
It's in the 3MF system. I suspect that if you loaded a .3MF file from an umlaut path it wouldn't load.

lordofhyphens, 2019-01-06:
@Samir55

Samir55, 2019-01-06:
I will take a look, LoH .

alranel, 2019-01-06:
If that's the case, make sure you use `boost::nowide` like we do for the other `IO::` classes

Samir55, 2019-01-24:
boost::nowide is used in reading the extracted files from 3mf but the problem is that miniz doesn't extract the file at the first place. I am looking for an easy fix.

3bhady, 2019-03-31:
@lordofhyphens @Samir55 
Problem isn't in miniz as I've tested it by itself and it can work with a string like this "Geräte" in it's path.
I think it's in the 23_3mf.t perl file I've fixed it by decoding the input path and output path to utf8 using `Encode::decode_utf8()` from `Encode`, and it's working properly.
But I'm not sure where else should I add this fix beside this test case as it will cause a problem every time `read_tmf` is called with special characters like 'ä'.
Considering this is a perl side issue it should be fixed before calling `Slic3r::IO::TMF::read(input_file, THIS)` which is in c++.