slic3r/Slic3r #4652: modifiers' extra features

Tinchus2009, 2018-12-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

This is a eature request. Other similar request might have been posted in the past, but some of them appear to be made in 2015... so I might have missed something?
We many time face model where supports are actually not necesary in some areas and in other they do. If we could have an extra choice to disable supports in some areas through the use of a modifier we could actually sole this. So this would be 1 of the feature request: to have the hability to activate/deactivate supports in the modifier's options

The other request: to improve overhangs we use coooling and speed. We can right now control the speed of an area of the model useing modifiers but we cant control the layer fan speed. Why would this be usefull?: if we have a powerfull layer fan, we dont need to use it at 100% all the time, speciually with filaments where too much cooling creates problems woth layer adhesion or even changes in colour (ike with PETg). So haveing the ability to just use more cooling in the overhangs areas would help a lot

supermerill, 2019-01-02:
> So haveing the ability to just use more cooling in the overhangs areas would help a lot

filament settings -> cooling -> bridges fan speed (it's used for bridges and overhangs)

lordofhyphens, 2019-01-06:
@Tinchus2009 please split this into 3 issues (so add 2 more), else I will fix one and close the issue.