slic3r/Slic3r #4654: Slic3r freezes during the brim calculation

hariseldon78, 2018-12-28:
**Describe the bug**
Slic3r freezes during the brim calculation of this model:
[pulley.zip](https://github.com/slic3r/Slic3r/files/2714889/pulley.zip)
, and the interface become irresponsive, even after cancelling the calculation. It happens with the object non rotated or rotated 90 on x axis. It slices fine without brim.

**To Reproduce**
Steps to reproduce the behavior:
1. add the attached model
2. configure a brim (2 mm in my configuration)
3. click on Preview
4. after a couple of seconds the slic3r crashes and the ui become irresponsive

**3D Model and Slic3r Configuration Export**
[config.zip](https://github.com/slic3r/Slic3r/files/2714893/config.zip)
[pulley.zip](https://github.com/slic3r/Slic3r/files/2714889/pulley.zip)

**Expected behavior**
correct slicing, the model is manifold and I didn't find any problem with it.

**Desktop (please complete the following information):**
 - Linux (Linux  4.15.0-43-generic #46-Ubuntu SMP Thu Dec 6 14:45:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux)

 - Version 1.3.0

lordofhyphens, 2018-12-31:
Traceback:
````
#0  0x00007f947a8fff11 in Slic3r::_align_to_grid (spacing=0, coord=-499899) at src/libslic3r/Point.cpp:324
324	    return aligned;
[Current thread is 1 (Thread 0x7f944f7fe700 (LWP 14111))]
(gdb) back
#0  0x00007f947a8fff11 in Slic3r::_align_to_grid (spacing=0, coord=-499899) at src/libslic3r/Point.cpp:324
#1  Slic3r::Point::align_to_grid (this=this@entry=0x7f944f7fd550, spacing=..., base=...) at src/libslic3r/Point.cpp:330
#2  0x00007f947a9206d0 in Slic3r::FillRectilinear::_fill_single_direction (this=0x7f93b3aa8b90, expolygon=..., direction=..., x_shift=<optimized out>, out=0x7f944f7fd770) at src/libslic3r/Fill/../Point.hpp:36
#3  0x00007f947a92294b in Slic3r::FillRectilinear::_fill_surface_single (this=this@entry=0x7f93b3aa8b90, thickness_layers=<optimized out>, direction=..., expolygon=..., out=out@entry=0x7f944f7fd770)
    at /usr/include/c++/8/bits/stl_iterator.h:804
#4  0x00007f947a958b49 in Slic3r::Fill::fill_surface (this=0x7f93b3aa8b90, surface=...) at /usr/include/c++/8/bits/stl_vector.h:930
#5  0x00007f947a90875a in Slic3r::Print::_make_brim (this=0x561d902773e0) at /usr/include/c++/8/bits/stl_uninitialized.h:115
#6  0x00007f947a602e1f in XS_Slic3r__Print__make_brim (my_perl=<optimized out>, cv=<optimized out>) at /usr/bin/perl -MExtUtils::XSpp::Cmd -e xspp -- -t "../xsp/typemap.xspt"  "../xsp/Print.xsp":1234
#7  0x0000561d8c479d11 in Perl_pp_entersub ()
#8  0x0000561d8c470026 in Perl_runops_standard ()
#9  0x0000561d8c3e3f02 in Perl_call_sv ()
#10 0x00007f947b0e852e in S_jmpenv_run (my_perl=my_perl@entry=0x561d915ed360, action=action@entry=0, thread=thread@entry=0x561d90d57c50, len_p=len_p@entry=0x7f944f7fdebc, 
    exit_app_p=exit_app_p@entry=0x7f944f7fdeb4, exit_code_p=exit_code_p@entry=0x7f944f7fdeb8) at threads.xs:508
#11 0x00007f947b0eaf7d in S_ithread_run (arg=0x561d90d57c50) at threads.xs:602
#12 0x00007f947b2f8fa3 in start_thread (arg=<optimized out>) at pthread_create.c:486
#13 0x00007f947b22988f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
````

Interesting, the coordinate is strongly negative.

lordofhyphens, 2018-12-31:
The negative coordinate isn't anything interesting. However, the spacing is, in this case, 0. I'm not sure why currently.

hariseldon78, 2018-12-31:
the stl is generated in freecad, i can post the project file tomorrow if it can help debugging

lordofhyphens, 2018-12-31:
@hariseldon78 no, the model as posted is enough to trigger the crash and I found at least one of the pieces.

So it looks like the trigger was the 125% first layer setting. That, plus the brim connections, is causing the brim code to bug out and try to produce a 0 spacing path (which is normally illegal). We have some asserts in the code but they aren't triggering in this situation (and I'm not sure why right now).

lordofhyphens, 2018-12-31:
@hariseldon78 With that model, a current workaround is to either set the brim connections to 0mm or set the first layer width to 200% (the default).

There's something going on with I think the flow code (remember, % width is relative to the _layer height_) that when Slic3r is trying to place that brim connector, the spacing is 0. 

The spacing value (which is the width-to-width) is being set incorrectly with that brim connection with and the math happens to work out that it comes out to be 0 on that model and a line width of ~0.5mm.

lordofhyphens, 2018-12-31:
Well, I know why it's tripping the assert: the build didn't have SLIC3R_DEBUG set, which automatically turns off asserts.

lordofhyphens, 2018-12-31:
@hariseldon78 in about 30-40 minutes you can check https://dl.slic3r.org/dev/linux/branches/ for a build of PR4662 that should sort out the crash. Please let me know if it fixes it for you (it does when I was testing it with your provided config and model).

What was happening is that even with the 0.5 fudge factor, with a 125% and 0.4 first layer height it happens to not reach the threshold for the math and gets rounded down to 0; 0 spacing breaks almost all of the assumptions in the slicer. I've patched a backup approach into the flow calculation that tries to choose an "okay" flow calculation in these situations (which I'm fairly certain only occur with brims).

lordofhyphens, 2018-12-31:
@hariseldon78 https://dl.slic3r.org/dev/linux/branches/Slic3r-1.3.1-dev-10f1d0b-PR4662-x86_64.AppImage

PR #4662 (2018-12-31): Fix edge case in spacing calculation for solid spacing.
In some situations the calculated solid spacing can be 0 with first-layer  widths of ~0.41mm and a requested brim connection width of 1mm, even though the actual spacing is nonzero.

Add a test case to flow test suite to cover this situation.

Fixes #4654
closed 2018-12-31 by lordofhyphens