slic3r/Slic3r #4655: Variable speed linked to adaptive slicing

howroyd, 2018-12-28:
Feature request:
I'm always frustrated when a print fails or prints badly around the complex detail of a design; usually where adaptive slicing has noticed and reduced the layer height.  Example attached.

**Describe the solution you'd like**
It would be great if you could reduce the speed with respect to the layer height so that the complex parts print slower and the simple bits print faster.

**Describe alternatives you've considered**
Manually changing the speed on the printer during the print.

**Additional context**
Loving Slic3r, thank you so much for a champion bit of software

[image]
.

lordofhyphens, 2018-12-31:
Have you played around with Autospeed at all? It tries to maintain a constant flow rate.

supermerill, 2019-01-03:
Shouldn't autospeed do the opposite of the request?.

lordofhyphens, 2019-01-06:
@howroyd can you post a 3D model where you are seeing bad results and the accompanying config?

howroyd, 2019-01-06:
Yes, autospeed would do the opposite so I have not tried it.

I can try and find a good example and post as requested.  Might take some time as I want to be certain that whatever I post is not being affected by any other variables, which of course, isn't trivial.  It is something I noticed particularly with HIPS.