slic3r/Slic3r #4657: customized colors in colorpint preview

wnc2, 2018-12-29:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
i wish to set custom colors in colorprint preview to have a more realistic render

[image]

**Additional context**
Add any other context or screenshots about the feature request here.

lordofhyphens, 2018-12-29:
Already there. View->"Color toolpaths by filament"

lordofhyphens, 2018-12-29:
Also, this isn't Prusa3d slic3r issue tracker.

closed 2018-12-29 by lordofhyphens

lordofhyphens, 2018-12-30:
@bubnikv I've put the notice and a link to your tracker in the default template in big text to hopefully catch more of these. 

@wnc2 When moving to the separate feature and bug report templates, I forgot to add the notice (that people only read 15% of the time) to not report Prusa3D specific issues here. I've fixed that.

wnc2, 2018-12-30:
Tried to be collaborative showing 2 feature request and a bug, I though slic3r and PU version were the same family

Regards

lordofhyphens, 2018-12-31:
This feature request looks like it is already what you want, the bug needs to be reproducible on our code (prusa fork has diverged a lot to support their printers first and foremost).

lordofhyphens, 2018-12-31:
And the other feature request (regarding plater window visibility) is already done in Slic3r (just change a setting in Preferences to use dialogs instead of tabs and turn background processing on).

As far as inapplicable issues go, these weren't bad. I have seen people open "pull requests" from prusa3d fork to us and make a bug report against prusa3d!

Obviously if a bug is reproducible on a build from dl.slic3r.org, then I do want to hear about it. If it isn't, then there is nothing I can do but point you at the right place and close the issue here.