slic3r/Slic3r #4663: Slic3r fails to load GUI: Slic3r-1.3.1-dev-26b2a1e-x86_64.AppImage

lordofhyphens, 2018-12-31:
@mderouss wrote the following:
 
> Attempting to start Slic3r-1.3.1-dev-26b2a1e-x86_64.AppImage ( dated 2018-11-10 19:17 ) in Xubuntu 18.04 ( fully patched ) with freeglut3 v 2.8.1-3 installed, results in the following terminal output -
> 
> libGL error: unable to load driver: nouveau_dri.so
> libGL error: driver pointer missing
> libGL error: failed to load driver: nouveau
> libGL error: unable to load driver: swrast_dri.so
> libGL error: failed to load driver: swrast
> The program 'slic3r.pl' received an X Window System error.
> This probably reflects a bug in the program.
> The error was 'BadValue (integer parameter out of range for operation)'.
> (Details: serial 569 error_code 2 request_code 155 minor_code 24)
> (Note to programmers: normally, X errors are reported asynchronously;
> that is, you will receive the error a while after causing it.
> To debug your program, run it with the --sync command line
> option to change this behavior. You can then get a meaningful
> backtrace from your debugger if you break on the gdk_x_error() function.)
> 
> This problem also occurs with Slic3r-1.3.0-dev-435fa6f-x86_64.AppImage.
> The system is using the nouveau drivers for an NVidia Quadro 4000 card. Other AppImages work fine, including the Prusa version of Slic3r ( Slic3rPE-1.41.2+linux64-full-201811221508.AppImage ).

lordofhyphens, 2018-12-31:
Posted this for @mderouss because apparently it got written up at https://github.com/openscad/openscad/issues/2614 which was closed as "wrong place for this". I only noticed because @probonopd responded and I checked the activity for @mderouss. 

Tagging @probonopd as he commented in #4042

probonopd, 2018-12-31:
Learned a new word today,

> necromancy

:+1:

lordofhyphens, 2018-12-31:
@mderouss  I can't reproduce with https://dl.slic3r.org/dev/linux/Slic3r-1.3.1-dev-8e5a9cb-x86_64.AppImage on a freshly installed Xubuntu 18.04.

Please try again and let me know if this works (I made some changes to the packaging system a few days ago).

lordofhyphens, 2019-01-01:
@mderouss https://dl.slic3r.org/dev/linux/branches/Slic3r-1.3.1-dev-4ff81a6-PR4666-x86_64.AppImage Can you give that a try?

probonopd, 2019-01-03:
Same as #4645?

lordofhyphens, 2019-01-03:
@probonopd I believe so.

prof7bit, 2019-01-04:
preloading the standard library makes it work:

LD_PRELOAD='/usr/lib/x86_64-linux-gnu/libstdc++.so.6' ./Slic3r-1.3.1-dev-f487129-x86_64.AppImage

As mentioned in the following link the bundled standard library is out of date:
https://superuser.com/questions/1377555/slic3r-appimage-fails-to-run-with-libgl-error-unable-to-load-driver-radeonsi/1377556

probonopd, 2019-01-04:
In this case, https://github.com/AppImage/AppImageKit/wiki/Creating-AppImages#libstdcso6 may apply, and we may need to use something like https://github.com/darealshinji/AppImageKit-checkrt/.

lordofhyphens, 2019-01-04:
@probonopd I am using the script version of checkrt now :/

It does work on my Debian Sid system (but I use the Nvidia drivers, not radeon).

probonopd, 2019-02-24:
I can reproduce this.

```
me@host:~$ '/home/me/Downloads/Slic3r-1.3.0-x86_64.AppImage'
libGL error: unable to load driver: r600_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: r600
libGL error: unable to load driver: r600_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: r600
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 1192 error_code 2 request_code 156 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

me@host:~$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="18.04 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```

The Slic3r PE AppImage does not suffer from this.

probonopd, 2019-03-02:
If you remove `./usr/lib/libstdc++.so.6'` from the AppImage, then it works.

Proof:

```
chmod +x ./Slic3r-1.3.0-x86_64.AppImage
./Slic3r-1.3.0-x86_64.AppImage --appimage-extract
wget https://raw.githubusercontent.com/AppImage/pkg2appimage/master/functions.sh
. ./functions.sh 
cd squashfs-root/
./AppRun # FAILS
delete_blacklisted 
./AppRun # WORKS
```

probonopd, 2019-03-02:
Note that this has resolved itself in the latest development builds, e.g., `Slic3r-1.3.1-dev-05a2f90-x86_64.AppImage`, thanks to

https://github.com/slic3r/Slic3r/blob/05a2f9028a35c56e039cf7a531d63b7ff7e97645/package/linux/appimage.sh#L53-L63

closed 2019-03-02 by lordofhyphens

probonopd, 2019-06-11:
Please reopen.

This issue seems to be back.

Fails to run on `xubuntu-18.04-desktop-amd64.iso`:

```
me@host:~$ /isodevice/Applications/Slic3r-1.3.0-x86_64.AppImage 
libGL error: unable to load driver: r600_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: r600
libGL error: unable to load driver: r600_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: r600
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 1192 error_code 2 request_code 156 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

me@host:~$ /isodevice/Applications/Slic3r-1.3.1-dev-48d5b11-x86_64.AppImage 
libGL error: unable to load driver: r600_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: r600
libGL error: unable to load driver: r600_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: r600
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 575 error_code 2 request_code 156 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```

The PrusaSlicer AppImage is working on the same system.

lordofhyphens, 2019-06-11:
PrusaSlicer working means very little at this point, but please open a new issue so we can mark it as regression.

probonopd, 2019-06-11:
I have opened https://github.com/slic3r/Slic3r/issues/4823.