slic3r/Slic3r #4664: Slic3er core dumps on Ubuntu 18.10

mrvanes, 2018-12-31:
After fixing the absence of perl-local in #4635 Slic3r now Core dumps on my Ubuntu 18.10 (Cosmic) system:
```
$ ./Slic3r 
Aborted (core dumped)
```
I use https://dl.slic3r.org/dev/linux/Slic3r-master-latest.tar.bz2

lordofhyphens, 2018-12-31:
As a test, rename ~/.Slic3r directory and try again

lordofhyphens, 2018-12-31:
I suspect that it is crashing on loading one of your print configurations. I had this happen on my system but was unable to reproduce if I built it on my development system. 

I will look into it a bit more and see what could be causing the problem; I just realized how I could narrow down the troublemaker.

lordofhyphens, 2018-12-31:
When reproducing, you can get a better backtrace by editing ./Slic3r and replacing line 6 with:
````
exec "gdb" --args "$DIR/perl-local" -I"$DIR/local-lib/lib/perl5" "$DIR/slic3r.pl" $@
````

Run it and hit 'r' followed by enter.
Then you can enter 'bt full' and paste the results here. It will tell me at least where it's crashing.

lordofhyphens, 2018-12-31:
It's an artifact of the build system; the travis build system was using gcc-4.9 as its base compiler forever. 

Unfortunately, we apparently ran into some weird situation where it does not like having std::strings as an argument to an exception class. That's my best guess at this moment.

What you are almost certainly running into are unknown options in your configuration file (which throws an exception now). The act of throwing the exception throws another (uncaught) exception for whatever reason.

lordofhyphens, 2019-01-01:
@alranel it looks like the exception classes are somehow causing this specific crash. My backtraces in gdb point right at the throw statement for invalid option exceptions.

I thought it was specific to GCC49 as I cannot reproduce on my dev system (uses gcc8.2)

I have installed a Trusty VM that should be close in setup to how Travis has its images configured. I have not had much luck getting a portable build out if the system, unfortunately.

mrvanes, 2019-01-01:
Renaming the ~/.Slic3r works around the problem, but makes me loose my profiles of course.
The backtrace as requested:
```
$ ./Slic3r 
GNU gdb (Ubuntu 8.2-0ubuntu1) 8.2
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/martin/Downloads/Slic3r/Slic3r/perl-local...done.
(gdb) r
Starting program: /home/martin/Downloads/Slic3r/Slic3r/perl-local -I/home/martin/Downloads/Slic3r/Slic3r/local-lib/lib/perl5 /home/martin/Downloads/Slic3r/Slic3r/slic3r.pl
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffeb9e0700 (LWP 4079)]
[New Thread 0x7fffe9cd2700 (LWP 4080)]
[New Thread 0x7fffe94d1700 (LWP 4081)]

Thread 1 "perl-local" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt full
#0  0x00007ffff7bf0077 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
        set = {__val = {0, 0, 0, 0, 4616187858836153959, 0, 55145056, 0, 18446462598732840961, 18446462598732840960, 4294967295, 1095216660480, 140737353789864, 23729792, 59418704, 61876968}}
        pid = <optimized out>
        tid = <optimized out>
#1  0x00007ffff7bd1535 in __GI_abort () at abort.c:79
        save_stage = 1
        act = 
          {__sigaction_handler = {sa_handler = 0x7ffff7ffe190, sa_sigaction = 0x7ffff7ffe190}, sa_mask = {__val = {140737351033907, 0, 1, 140737353859056, 140737488342224, 0, 140737354131240, 4194368, 140737488289801, 131, 0, 0, 0, 0, 7405853560218348544, 140737488343864}}, sa_flags = -12176, sa_restorer = 0x7fffffffd070}
        sigs = {__val = {32, 0 <repeats 15 times>}}
#2  0x00007ffff62713be in _Unwind_SetGR () at /home/martin/Downloads/Slic3r/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#3  0x00007ffff621b3a1 in __gxx_personality_v0 () at /home/martin/Downloads/Slic3r/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#4  0x00007ffff7fbad73 in  () at /lib/x86_64-linux-gnu/libgcc_s.so.1
#5  0x00007ffff7fbb2d1 in _Unwind_RaiseException () at /lib/x86_64-linux-gnu/libgcc_s.so.1
#6  0x00007ffff3690d07 in __cxa_throw () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff5fe8212 in Slic3r::ConfigBase::set_deserialize(std::string, std::string, bool) () at /home/martin/Downloads/Slic3r/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#8  0x00007ffff5fed7dd in Slic3r::ConfigBase::load(std::string const&) () at /home/martin/Downloads/Slic3r/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#9  0x00007ffff5f0adc7 in XS_Slic3r__Config__load(interpreter*, cv*) () at /home/martin/Downloads/Slic3r/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#10 0x00000000004c0050 in Perl_pp_entersub ()
#11 0x00000000004b8736 in Perl_runops_standard ()
#12 0x000000000043af62 in Perl_call_sv ()
#13 0x00007ffff4a399bf in XS_Wx___App_Start(interpreter*, cv*) () at /home/martin/Downloads/Slic3r/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Wx/Wx.so
#14 0x00000000004c0050 in Perl_pp_entersub ()
#15 0x00000000004b8736 in Perl_runops_standard ()
#16 0x0000000000442889 in perl_run ()
#17 0x000000000041cb9b in main ()
```

lordofhyphens, 2019-01-01:
Wow, what a pain this was.

@mrvanes does this work on your system? I was able to get it to run on Xubuntu 18.04

https://dl.slic3r.org/dev/linux/branches/Slic3r-1.3.1-dev-4ff81a6-PR4666-x86_64.AppImage

closed 2019-01-01 by lordofhyphens

mrvanes, 2019-01-02:
Both Slic3r-1.3.1-dev-4ff81a6-PR4666-x86_64.appimage and slic3r-1.3.1-dev-45353b7-linux-x64.tar.bz2 now work for me again. Thx for the hard work!!

PR #4666 (2019-01-01): Use GCC8 instead of 4.9 for travis
Update travis build items to use gcc8 instead of gcc4.9

Fixes #4664