slic3r/Slic3r #4665: Building Slic3r on OS X with macports boost fails due to boost libraries not found

Blokkendoos, 2018-12-31:
**Describe the bug**
Building Slic3r on OS X with macports boost fails due to boost libraries not found.

**To Reproduce**
This is needed for the script to find wxperl_overload:
$ sudo ln -s /opt/local/bin/wxperl_overload-5.26 /opt/local/bin/wxperl_overload

Steps to reproduce the behavior:
1. git clone https://github.com/slic3r/Slic3r.git
2. cd Slic3r
3. sudo SLIC3R_STATIC=1 BOOST_INCLUDEPATH=/opt/local/include BOOST_LIBRARYPATH=/opt/local/lib perl Build.PL --gui
4. sudo SLIC3R_STATIC=1 BOOST_INCLUDEPATH=/opt/local/include BOOST_LIBRARYPATH=/opt/local/lib perl Build.PL
5. Results in:
Configuring /Users/johan/Desktop/Slic3r/xs ... Slic3r requires the Boost libraries. Please make sure they are installed.

If they are installed, this script should be able to locate them in several
standard locations. If this is not the case, you might want to supply their 
path through the BOOST_INCLUDEPATH and BOOST_LIBRARYPATH environment variables:

    BOOST_INCLUDEPATH=/usr/local/include BOOST_LIBRARYPATH=/usr/lib perl Build.PL

If you just compiled Boost in its source directory without installing it in the
system you can just provide the BOOST_DIR variable pointing to that directory.

N/A
! Configure failed for ./xs. See /Users/johan/.cpanm/work/1546259450.10165/build.log for details.
The XS/C++ code failed to compile, aborting
$

**Expected behavior**
Succesfull build

**Work-around**
1. vi xs/Build.PL
...
    #my $lib_ext = $Config{lib_ext};
    my $lib_ext = '.dylib';
:wq
2. sudo SLIC3R_STATIC=1 BOOST_INCLUDEPATH=/opt/local/include 
[2018-12 Build Slic3r FAIL en OK.txt](https://github.com/slic3r/Slic3r/files/2718334/2018-12.Build.Slic3r.FAIL.en.OK.txt)
BOOST_LIBRARYPATH=/opt/local/lib perl Build.PL
3. Results in a successful build
4. perl slic3r.pl => Ok

**Desktop (please complete the following information):**
 - OS: MacOS 10.10.5 (Yosemite)
 - Version 1.3.0-1270-gf19fc3b6

**Additional context**
Perl5.26 and Boost 1.66.0 installed w Macports 2.5.4.

lordofhyphens, 2018-12-31:
Thank you for the workaround.

Why is the modification to lib_ext necessary?

Could you write this up on the Slic3r wiki? I don't see this as something that will drive a build environment change.

Blokkendoos, 2018-12-31:
The build script derives the boost lib extension from the boost system lib (if I remember well). In my (macports) installation some boost libs have both a static and a dynamic version, e.g. libboost_system, the others only dynamic. See file list below. Maybe this is due to the macports installation (port install boost).
I will add the workaround to the wiki.
```
$ port installed boost
Warning: port definitions are more than two weeks old, consider updating them by running 'port selfupdate'.
The following ports are currently installed:
  boost @1.66.0_2+no_single+no_static+python27
  boost @1.66.0_3+no_single+no_static+python27 (active)
```

```
$ ls -la /opt/local/lib | grep boost | grep system
-rwxr-xr-x     1 root      admin     131844 20 mrt  2018 libboost_filesystem-mt.dylib
-rw-r--r--     1 root      admin     106048 20 mrt  2018 libboost_system-mt.a
-rwxr-xr-x     1 root      admin      27692 20 mrt  2018 libboost_system-mt.dylib

ls -la /opt/local/lib | grep boost | head
-rwxr-xr-x     1 root      admin      10232 20 mrt  2018 libboost_atomic-mt.dylib
-rw-r--r--     1 root      admin     444408 20 mrt  2018 libboost_chrono-mt.a
-rwxr-xr-x     1 root      admin      41552 20 mrt  2018 libboost_chrono-mt.dylib
-rwxr-xr-x     1 root      admin     111320 20 mrt  2018 libboost_container-mt.dylib
-rwxr-xr-x     1 root      admin      18460 20 mrt  2018 libboost_context-mt.dylib
-rwxr-xr-x     1 root      admin      68256 20 mrt  2018 libboost_coroutine-mt.dylib
-rwxr-xr-x     1 root      admin      75064 20 mrt  2018 libboost_date_time-mt.dylib
-rw-r--r--     1 root      admin       4040 20 mrt  2018 libboost_exception-mt.a
-rwxr-xr-x     1 root      admin     131844 20 mrt  2018 libboost_filesystem-mt.dylib
-rwxr-xr-x     1 root      admin     498644 20 mrt  2018 libboost_graph-mt.dylib
```