slic3r/Slic3r #4673: New placeholders min_x, max_x, min_y, max_y

danielkucera, 2019-01-03:
**Is your feature request related to a problem? Please describe.**
My printbed is not completely flat. I use auto bed leveling but that's not always perfect.

**Describe the solution you'd like**
I'd like to modify the start code to issue the G29 command with L,R,F,B parameters (http://marlinfw.org/docs/gcode/G029-abl.html) to speed up the leveling and get more suitable bed mesh specific for a print. These parameters could be set using placeholders so only thing needed is to define new placeholders holding the value of min_x, max_x, min_y, max_y.  

**Describe alternatives you've considered**
No idea.

**Additional context**
None.

lordofhyphens, 2019-01-03:
Since this wouldn't be used anywhere else in Slic3r and you would have to edit the printer config anyway, I don't see value in adding new config options (which is what these placeholders are) when it is simpler to inline the values.

lordofhyphens, 2019-01-03:
https://manual.slic3r.org/expert-mode/printer-settings

danielkucera, 2019-01-03:
Hi @lordofhyphens , you might have misunderstood me. These min/max values refer to size of the model, not to size of the printbed. I think it makes more sense to level only the portion of printbed which will be used for printing so when I print 10x10x10 mm cube, I don't need to level 250x250mm of printbed.

lordofhyphens, 2019-01-03:
So you want gcode placeholders for the bounding box covering all models on the plate?

danielkucera, 2019-01-03:
exactly

lordofhyphens, 2019-01-04:
Sorry, you would see it if you compiled the c++ cli. Because support hasn't been ported yet, I have not turned on that code path from the GUI.

There is a corresponding code path in the perl side of things.

lordofhyphens, 2019-01-04:
Did you add your new options to PrintConfig?

danielkucera, 2019-01-04:
No.