slic3r/Slic3r #4682: layer height not recalculated when you switch configs

djdelorie, 2019-01-06:
Switching between configs which specify different layer heights, does not cause the layer heights to be recalculated.

**To Reproduce**
Load the two configs, config-med and config-fine.  Start slic3r and select the fine config.  Load test cube.
Note that the preview shows layers 0.1 0.2 0.3 etc.  Switch to the config-med config in "print settings".  Note that the preview *still* has 0.1 0.2 0.3 when it should have 0.2 0.4 0.6.
Manually changing the layer height in the print settings dialog does work.

Note that I have background processing enabled in my preferences.

**3D Model and Slic3r Configuration Export**
[bug-layers.zip](https://github.com/slic3r/Slic3r/files/2730984/bug-layers.zip)

**Expected behavior**
Changing configs should recalculate layers.

**Desktop (please complete the following information):**
Fedora 26
slic3r master: 1.3.0-1281-gdabb5f38

lordofhyphens, 2019-01-09:
Looks like the configuration change either isn't invalidating the config or it's not having the desired effect. I don't know which right now.

lordofhyphens, 2019-01-09:
Since it's recalculating if you change the specific config, I suspect it's in the UI.