slic3r/Slic3r #4687: New feature: brim "ear"

supermerill, 2019-01-09:
Hi

When i embed "ears" in my model before printing, it's very hard to remove, because the printing lines aren't along the perimeter of the "real" object (especially if the bottom layer is rectilinear). 

So i quickly created an option to create a brim only over "sharp" convex corners.

currently 2 new parameters: 

* ear brim (boolean)
* max angle(float, default to 120)

I need  to do a bit more work to improve the ordering of the printed gcode.

[image]

What do you think of it?
Shoud i do a pr?

VanessaE, 2019-01-09:
Definitely make a PR of this!  (though I'd suggest making each ear from a continuous bead, rather than multiple parallel lines as seen here)

supermerill, 2019-01-09:
[image]

UticaTechClub, 2019-01-10:
Very good idea.
Brim the most susceptible to bad adhesion areas.

Correct me if I am wrong...
Implementation can be tricky on models with many sharp corners. Gears come to my mind.

lordofhyphens, 2019-01-10:
Because of the way it is generated, I bet it would devolve to a standard brim in the gear situation.

supermerill, 2019-01-10:
yes, it's only a mask + some post-process over the current (external) brim.

closed 2019-01-11 by lordofhyphens

Splarv, 2019-02-05:
Eh, I am experimenting to use PETG without hot plate ant clue. And I heavely need this. The reason is when you type continuous brim, after cooling the plastic begin shrink and brim too. And brim too begin pull corners to each other to warp corners. What I see, the wider the brim, the worth result and brim get out of plate first and begin pull corners later. And I think the "brim ears" is opposing thing. Ears will only pull corners in the opposite direction and this will bring positive and desirable effect.