slic3r/Slic3r #4692: Brim ear don't work well with sharp-rounded corners

gege2b, 2019-01-11:
Hi there and happy new year everyone

**Describe the bug**
The new great brim ears feature seems to only measure sharp angles (witch makes sense)
But on models with lightly rounded but still "globally" sharp corners, the ears aren't generated
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. open a model with sharp-rounded angles (see above for the link)
2. activate the brim ear with default values (I just spotted a typo here, I will open a new ticket for this)

**3D Model and Slic3r Configuration Export**
i.e. [this one](https://www.thingiverse.com/download:1488247) (look at the tail), other config are unrelated

**Expected behavior**
getting an ear on the tail of the crocodile, obviously, this zone has a very high probability to warp

** Screenshots **
With default ears settings
[image]

Need to bump up the ear angle to 176 degree to have the brim on the tail. BTW The result is weird and a simple brim should be better in this case
[image]

**Desktop (please complete the following information):**
 - OS: Win10
 - Version commit 3b338e5f84fafaaea32b9f8c11633aa231380285

supermerill, 2019-01-11:
yes, it's a problem.

But it's much more complicated to do. Instead of just picking convex corners (the function was already here), we have to:
* create a new psettign for the precision
* integrate the angle over a distance, for any points.
* or simplify drastically the polygon before to create sharp corners from round ones (if it works)

I'm not doing it for the near future, i have other things to do. If someone wants to patch it, do it!

ledvinap, 2019-01-11:
@supermerill : It should not be that hard:

- precision may be taken relative to extrusion width, probably something between 0.5 and 1, not user configurable

- for each tested point find point that is precision clockwise and counterclockwise on contour.

- use these points to calculate angle

ledvinap, 2019-01-11:
And it may be necessary to simplify / remove points that are too close to keep reasonable performance in ears intersection

Splarv, 2019-02-05:
I think just option to manually switch on continuous brim, when it necessary (for instance for models with rounded corners) is enough. :) Or there can be solution to manually place ear brims, but this is another feature.