slic3r/Slic3r #4693: Typo on the tooltip of the shortcut for brim ear angle

gege2b, 2019-01-11:
It's so tiny that I would have corrected this myself but I didn't found where the erroneous value was 😭 

**Describe the bug**
The default value show on the tooltip for the shortcuts for brim ear angle is 90°
 Tooltip on the settings dialog and the command line "--help" shows 125°
Default value is 125°

**To Reproduce**
Steps to reproduce the behavior:
1. Show the brim ear angle on the shortcut pane
2. hover the mouse and look at the tooltip

**3D Model and Slic3r Configuration Export**
not relevant

**Expected behavior**
Good value shown

**Screenshots**
If applicable, add screenshots to help explain your problem.
(look at the  tooltip)
[image]

**Desktop (please complete the following information):**
 - OS: Win10
 - Version 3b338e5f84fafaaea32b9f8c11633aa231380285

lordofhyphens, 2019-01-13:
Tooltip shortcuts are in xs/src/libslic3r/PrintConfig.cpp

lordofhyphens, 2019-01-13:
That being said, the "default" value shown in shortcut tooltips should be showing whatever is saved to the profile. So if you have 90 saved, than the shortcut default is 90. The "I have no config" value is 125.

closed 2019-01-13 by lordofhyphens