slic3r/Slic3r #4694: Change fan speed on toolchange

CrazyEngie, 2019-01-11:
### Version
1.3.0 64bit

### Operating system type + version
Windows 7 SP1 64bit

### Issue
How to set different fan speeds for different extruders?
Slic3r sets the fan speed only once, taking the value from "Filament settings" for first extruder.
I tried to use Custom G-code.
Here is my code from Toolchange G-code: M106 S[min_fan_speed_[next_extruder]]
In the ready code it looks like this: M106 S[min_fan_speed_0] and M106 S[min_fan_speed_1]
Is it even possible to change the fan speed in this way, and I just do not write the command correctly?
If not, then I think that automatically changing the blower for different extruders would be a very useful feature.

lordofhyphens, 2019-01-13:
I don't think the replacement gets chained like that.

lordofhyphens, 2019-01-13:
Probably would just be better to have toolchange gcode for filament (executed after printer) that has its own tool or something coded to it.