slic3r/Slic3r #4696: Slic3r 1.3.0 and 1.3.1 Linux don't seem to use multiple threads on Ryzen 8 core

ML3D, 2019-01-11:
Slicing a semi complex model, about 120mb STL, on my Ryzen workstation running Linux and the program doesn't seem to care if Threads in Preferences is set to 4, 8 or 16.

It took longer to preview this model than it does on an old Phenom CPU on Windows.
It is taking a LONG time. LIke 30 minutes or something. To the point where I thought it crashed. I loaded up multiple versions of Slic3r and tried it. When viewing system monitor, it shows only 1 core pegged at 100% and the rest chillin' around 0 or 1%.

Infill is Star pattern at 30%.
5 vertical shells, 4 bottom, 4 top layers. 

I tried with and without background processing. Didn't matter.

Let me know if you need more info.

lordofhyphens, 2019-01-12:
@ML3D are you building yourself or using a build from https://dl.slic3r.org?

Also, please provide a sample config/model.

ML3D, 2019-01-12:
I have only been using AppImages downloaded from Slic3r.org - tested release and dev versions.

I tried to upload the STL, but it is 66mb zipped. I'll try to find another place to upload it and link back.
[Slic3r_config_bundle.zip](https://github.com/slic3r/Slic3r/files/2752438/Slic3r_config_bundle.zip)

lordofhyphens, 2019-01-12:
@ML3D Please upload a Ctrl-E export (File->Export Config), not a bundle. Bundles mean that people testing have to a) wreck their own slic3r config directories and b) guess at what exact combination of settings. 

One other thing would be helpful; could you post a screenshot of File->Preferences. Here's mine for reference:
[image]

ML3D, 2019-01-12:
As requested:

[slic3r_config.zip](https://github.com/slic3r/Slic3r/files/2752450/slic3r_config.zip)
[image]

lordofhyphens, 2019-01-12:
With that config on my CPU (on Linux), I noticed a spike in the CPU usage, then it dropped down to 100%, then spun up again to lots.

I'm wondering if with those models there's a few bottlenecks in slicing (so the other threads stop work because they're done).

The parallelization strategy that Slic3r employs is to use a thread per-layer (up to maximum in settings). 

````
processor	: 15
vendor_id	: AuthenticAMD
cpu family	: 23
model		: 1
model name	: AMD Ryzen 7 1700 Eight-Core Processor
stepping	: 1
microcode	: 0x8001126
cpu MHz		: 1898.701
cache size	: 512 KB
physical id	: 0
siblings	: 16
core id		: 7
cpu cores	: 8
apicid		: 15
initial apicid	: 15
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb hw_pstate sme ssbd sev vmmcall fsgsbase bmi1 avx2 smep bmi2 rdseed adx smap clflushopt sha_ni xsaveopt xsavec xgetbv1 xsaves clzero irperf xsaveerptr arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif overflow_recov succor smca
bugs		: sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
bogomips	: 5987.87
TLB size	: 2560 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 43 bits physical, 48 bits virtual
power management: ts ttp tm hwpstate eff_freq_ro [13] [14]
````

ML3D, 2019-01-12:
Maybe this screen grab will help. I'm slicing again right now but it's still 'infilling layers' 20 minutes later.
I'm timing it and will report back when it's complete.
[image]

ML3D, 2019-01-12:
It finished in approx 22 minutes. Does that seem right? Am I just being impatient?

Speaking of processing, it would be pretty cool if Slic3r could process via CUDA. That would smoke.

lordofhyphens, 2019-01-12:
@ML3D Someone willing to port part of the c++ slicing engine to CUDA (or probably Vulkan or OpenCL) would be helpful to getting that done. I don't have time.

lordofhyphens, 2019-01-12:
@ML3D can you zip up the 3d model and attach it as well for testing?

ML3D, 2019-01-12:
I tried, too big. I'm working on uploading it somewhere for you to grab it.

lordofhyphens, 2019-01-12:
For that single X carriage? Weird.

I've been able to nail the area of 100% down to the prepare_infill method in PrintObject.cpp.

ML3D, 2019-01-12:
Try this. Ignore the config inside the zip.
https://file.io/dWLfNE

ML3D, 2019-01-12:
Maybe it's a combination of the model being too dense and 5 perimeters?

ML3D, 2019-01-12:
Well, I exported it again from the original application and I got the size down from 100+ mb to about 14mb. I was then able to slice it in only 2-3 minutes on the phenom with Win 8. I need to test this model now on Linux / Ryzen. I suspect it will be equally as fast. I think the density of the model was excessive.

lordofhyphens, 2019-01-12:
Yeah, I think there's a few serial sections in make_perimeters that get affected by an absurdly detailed model. 

The deprecated option "Resolution" prefilters the model (and I bet in your use case, would have helped).

ML3D, 2019-01-12:
Thanks for your work, help and feedback.

lordofhyphens, 2019-01-13:
For people looking into this or running into similar issues, there are some processes of the `make_perimeters()` code that run serially (so far as I can tell).

Samir55, 2019-01-24:
@lordofhyphens I am willing to work on that (using CUDA in slicing engine), I will talk with you about in the near time.

lordofhyphens, 2019-01-24:
@Samir55 I think we should be looking at OpenCL or Vulkan instead of CUDA myself, if only to avoid the hard dependency on nVidia cards (because as much as I generally like CUDA, doing that will basically be code duplication).