slic3r/Slic3r #4697: Control panel features

coosadog, 2019-01-12:
### Do not request features for the Prusa3D fork of Slic3r here without confirming it is applicable to the mainline Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

The progress meter needs to show elapsed time, somewhere.  Also, it would be nice to show the progress in a 3D image as in Pronterface.  I love having the control, now, though!  Thanks!

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

lordofhyphens, 2019-01-12:
Progress meter in which UI element?

coosadog, 2019-01-12:
The progress meter in the Controller panel when printing something.  It would be nice to know how much time is remaining for the print.  So, even if it was just elapsed time, it would help, because it already shows you the total amount of time for the print.  Thanks for your reply!  I know more will be added here.  I just wanted to put my 2 cents into it.  There's enough room for the 3D progress display, too....

VanessaE, 2019-01-12:
related:  https://github.com/slic3r/Slic3r/issues/4636