slic3r/Slic3r #4699: Plater does not work at all.

jammi, 2019-01-13:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

The plater doesn't render, moving the window adds random content from other windows on the system into the viewport.

**Desktop:**
 - OS: macOS Mojave, latest build, unlikely macOS version dependant
 - Version: All currently available 1.3.1-dev builds

jammi, 2019-01-13:
The previous build I used was from August 26:th last year and that seems to work ok.

jammi, 2019-01-13:
Screenshot of initial window right after starting Slic3r:

[image]

Screenshot after resizing the window (background of the window shows through)
[image]

lordofhyphens, 2019-01-13:
[image]

That *definitely* looks like broken/missing OpenGL (I've seen it on Windows before when they didn't have OGL libraries).

lordofhyphens, 2019-01-13:
I've re-enabled the older jenkins builds for the time being and pointed them at  https://dl.slic3r.org/dev/mac/jenkins/ as a workaround for Mojave users.

lordofhyphens, 2019-01-13:
Since the older builds work, there is something going on with the toolchain used on MacOS.

alranel, 2019-01-13:
We need to compare the two packages and see if anything is missing.

lordofhyphens, 2019-01-13:
The new packages do work on 10.13 (High Sierra). I have a new Mac Mini 2014 on its way.

lordofhyphens, 2019-01-18:
Well, I can reproduce on my "new" Mojave box. Time to get Slic3r bootstrapped.

lordofhyphens, 2019-01-20:
From-source builds on Mojave don't work either. Joy.

aleung, 2019-04-27:
@lordofhyphens The folder https://dl.slic3r.org/dev/mac/jenkins/ is empty. Where can I get the workaround?

lordofhyphens, 2019-04-27:
@aleung sorry about that, builds are up now.

lordofhyphens, 2019-12-06:
@aleung can you check https://mega.nz/#!8aoxhCbR!L-OMNmFp8K7h1og-1fzLAIn2DO3sPFui3TPN8Z8Dlvg for me when you have some time? I'm trying to figure out if it's just Travis being dumb.

aleung, 2019-12-08:
@lordofhyphens This one works on Mojave.