slic3r/Slic3r #4702: 3D Infill Patterns - wanting to reflect that in the support structure as well

JamesVerner, 2019-01-14:
Hey everyone,

Sorry to raise an issue, but trying to figure out if this is possible to do and if anyone might know how it can be done. This is due to the 3D infill patterns such as 3D honeycomb, gyroid and concentric. I noticed that with the infill patterns regarding these examples, it's dynamic and changes when you adjust the height with preview. 

[image]

However, when messing around and using it as a support structure, it doesn't seem to work in the same behaviour. But rather take a "snapshot" of it and use it as support.

[image]
 
I completely understand why it is the case, however, I'm just investigating if it was possible to make it as dynamic as similar to that for the infill patterns. Therefore would anyone know if this is possible at all currently and was disabled, or would this feature need to be implemented?

supermerill, 2019-01-14:
You may have to update Fill:z for each layer.

JamesVerner, 2019-01-14:
Sorry to ask, but whereabouts is the Fill::Z?

supermerill, 2019-01-14:
> when messing around and using it as a support structure

I suppose you changed something inside the code?

To change the support pattern, you changed the Fill class used?

You have to update the z parameter of the Fill instance before the code call the fill_surface method.

lordofhyphens, 2019-01-15:
Support infill use the same code as regular infill for generating the pattern, so far as I remember. 

You need to wire it up through PrintConfig.hpp and PrintConfig.cpp to get it to show up. There's some other differences in how it's applied, though, as @supermerill mentioned.

lordofhyphens, 2019-01-15:
[image]

I am not seeing much use, though, for the 3D patterns for support.

JamesVerner, 2019-01-16:
Hey guys, thanks for getting back to me and sorry for taking a day to get back to you both. I was able to figure out how to implement the 3D patterns dynamic due to the Fill::Z method. I will admit I can't see much use for it either since seems not needed, but just trying to see what could be done in general.