slic3r/Slic3r #4705: Infill offset when adaptive slicing layer height changes

mat813, 2019-01-15:
**Describe the bug**
When the layer height changes abruptly, the infill layers shifts, sometime ending up over empty air, fail to have any support at all.

**To Reproduce**
Use the provided model and config, render it.  Infill shifts between layers heights 38.31 and 38.96, and then 49.94 and 51.59. And it starts when the layer height changes, then stabilizes.

**3D Model and Slic3r Configuration Export**
[Archive.zip](https://github.com/slic3r/Slic3r/files/2761771/Archive.zip)

**Expected behavior**
The infill layers should stay aligned.

**Screenshots**
[image]
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version [e.g. 1.3.0] 916c30c

**Additional context**
I exported the config using the "stars" infill pattern, and added screenshots with stars and gyroid, the shift happens on all infill patterns.

Lucki, 2019-03-22:
I've also noticed this with honeycomb. The infill shifts on every adaptive layer height changes.

Desktop:

* OS: Arch Linux
* Version: 1.3.0

lordofhyphens, 2019-03-22:
Adaptive is probably breaking one of assumptions in that codebase, that or the line thickness is changing (which would rescale the polygon used).

supermerill, 2019-03-27:
On gyroid, it uses the z position and flow.spacing() to create the waves. I don't think there are something else.
Spacing change when the layer height change, because the half-circle is smaller. Maybe "width" should be used instead of "spacing"?

ctmakro, 2019-06-18:
can confirm this bug. Adaptive causes infill pattern to shift abruptly.

TD5023, 2019-09-29:
> Adaptive is probably breaking one of assumptions in that codebase, that or **the line thickness is changing (which would rescale the polygon used).**

I'll also confirm that the infill shifts on layer height changes, and that it's an issue across all infill patterns. Regarding line thickness variance, that's not the problem because the shift is still there with a constant thickness defined. Hopefully this is on someone's radar, because while it's not a huge deal for straight-line infills, I _really_ would like to be able to use gyroid and adaptive, but it's not feasible right now.

lordofhyphens, 2019-09-29:
It's because, as of the last time looked into it, the % setting was based on the layer height for that layer.

TD5023, 2019-09-29:
Yes, that's true if you use a % setting. However, it also comes with the option to set a constant thickness (e.g. 0.45 mm), so that should eliminate the need for spacial recalculations on non-solid infill. Even with a hard-defined thickness, the infill still shifts.

It should also be noted that all it does is shift. Even when a % is used, there is no change to the line density, which would be expected if the thickness were the issue.

lordofhyphens, 2019-09-30:
When I said %, I was explicitly referring to infill_density.

TD5023, 2019-09-30:
My mistake. I thought you were referring to line thickness calculations. I guess I always assumed infill density was based on a 2d snapshot of the layer, but if layer height is a factor and it's based on mass extruded, a slight change would make sense.