slic3r/Slic3r #4706: Variable Layer Height feature won't retain settings

jimmywizzle, 2019-01-17:
When adding or removing detail using the variable layer height sidebar in the "Plater", the applied changes reset/disappear as soon as the mouse button is released. 

**To Reproduce**
1. "Enable variable layer height" feature in "Printer Settings --> General"
2. Set "Layer height limits" in "Printer Settings --> Extruder1":
           eg.   Min = 0.05mm    Max = 0.3mm
3. Click "Layers Editing" icon in the "Plater" screen
4. In the variable layer height sidebar, left/right mouse click the line, then drag to add/remove detail.
    As soon as the mouse button is released, the changes reset to the original base setting.

**3D Model and Slic3r Configuration Export**
[STL_and_Config-Export.zip](https://github.com/slic3r/Slic3r/files/2770127/STL_and_Config-Export.zip)

**Expected behavior**
Variable layer height changes/settings are retained after mouse button is released. 

 - OS:   Windows 10
 - Version:   1.42.0 alpha 2 & 3

**Additional information**
The variable layer height feature worked as expected in v1.41.2, for the same model.

closed 2019-01-17 by jimmywizzle

lordofhyphens, 2019-01-17:
I see you closed it, can you reproduce the behavior with a build from dl.slic3r.org/dev

jimmywizzle, 2019-01-17:
Hi Joseph,
I do apologise; I'm quite new to Github.  After submitting this issue, I realised that I had posted it in the wrong place, hence why I closed it straight away.  My problem relates specifically to development build of Slic3r Prusa Edition.  Sorry for the inconvenience.
Kind regards,
James

lordofhyphens, 2019-01-18:
@jimmywizzle no worries; I am usually interested to know if people can reproduce in mainline Slic3r as well (because that indicates it's a bug that is in both repositories and should be fixed here :) )