slic3r/Slic3r #4709: Poke Yoke when you cut a piece into 2 half.

opensky727, 2019-01-21:
I bought a prusa MK3 last summer and I like it so much and I also, really like the Slic3r PE software, but I wanted to make a proposition for the next version of it. I would love that when you cut a part into two halves, one will have a peg and the other half a hole to be able to make an assembly quick and easy. Just add something in the settings to chose the correct distance between the peg and the hole and it would be just perfect! For more information read on the poke-yoke technique. Thanks and have a nice day!

lordofhyphens, 2019-01-21:
Probably wouldn't be a good idea to use a peg for your keys, but as 1.75mm hole on both sides (for inserting snippets of filament) or a round bump may be feasible.

Deforming the mesh in a useful way is much more complex than you'd like to think.

ledvinap, 2019-01-21:
@lordofhyphens : Printing peg would be almost impossible without support, but double hole is great idea.

In some cases one option is to print both parts side-by-side, with thin bridge (brim),  possibly indented in middle, that is bent to define piece position before glueing.

IMO this is task for mesh editing program, not slicer

lordofhyphens, 2019-01-21:
@ledvinap slic3r has some rudimentary mesh editing in its cut feature, which is the use case for this feature request.

Animo78, 2019-08-14:
Would love this request.
Now it takes a lot of troubles gluing it together.

James-G4CLF, 2019-12-24:
I support this request. If the shape of the cut is a circle it is very hard to align the two halves accurately - aligned holes would be a Godsend.