slic3r/Slic3r #4712: ADD ability to load Printer Settings → Start & End & Before/After layer change etc.. G-code from files

allanlaal, 2019-01-23:
please ability to load Printer Settings → Start & End & Before/After layer change etc.. G-code from files

3dprinter people are tinkerers by nature. I like to tinker with my start and end gcode continually
right now I copy paste the contents of these textareas to an editor, so I'll have more than 4 lines visibles at a time + I can version the changes with version control (mercurial). Version Control is important even here :)

jammi, 2019-01-25:
Since the configuration is in easily parseable text files, you could just add them to version control as is, and script a build support tool to attach the contents of your gcode files into the ini files. A real improvement however would be going from the proprietary .ini stuff to .yaml or .json config files.

lordofhyphens, 2019-01-25:
It's a simple INI format, older and simpler than JSON or YAML. I don't see a point in increasing the complexity of the system (through even more external dependencies) the reason of "well, it isn't web technology X".

https://en.m.wikipedia.org/wiki/INI_file

@allanlaal from the CLI, I can tell you offhand that if you use --load with a config ini that only has a few options entered, only those options get loaded. I am pretty sure you can chain loads.

There is nothing stopping you from editing the file externally (on Mac/Linux it is in ~/.Slic3r, for windows it is under appdata), just escape newlines.

I already keep my entire printer config folder under version control. 

https://github.com/lordofhyphens/printer-configurations