slic3r/Slic3r #4714: Missing interior brim.

mat813, 2019-01-23:
**Describe the bug**
The interior brim is not generated properly.

**To Reproduce**
Preview the stl with the config provided.

**3D Model and Slic3r Configuration Export**
[Archive.zip](https://github.com/slic3r/Slic3r/files/2788339/Archive.zip)

**Expected behavior**
There should be interior brim around all the "inside" stars.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version 916c30c

**Additional context**
May be related to #3993 but I am not sure.

lordofhyphens, 2019-01-23:
Are the interior stars part of the same model?

mat813, 2019-01-23:
Yes, the "stars" branches connect with each others and up to the sides.

mat813, 2019-01-23:
Like this:

[image]