slic3r/Slic3r #4715: 1.42.0alpha3: soluble interface is malformed

supermerill, 2019-01-24:
**Describe the bug**
using the support with soluble interface parameter, with 2 interface layers, on the 3 interface layers dawned, the bottom2 are ok (good simple rectangle) but the top one (the most important) has some " notches that totally deform it.

**To Reproduce**
1. load the .3mf
2. Click on slice
3. zoom on an interface layer (under a bridge)
4. See error

**3D Model and Slic3r Configuration Export**

**Expected behavior**
The 3 interface "polygon" should be the same.

**Screenshots**
[image]

**Desktop:**
 - Windows 64b

lordofhyphens, 2019-01-24:
Please reproduce on 1.3.0 or 1.3.1-dev

lordofhyphens, 2019-01-24:
This is showing up on main slic3r? I was pretty sure there was no code overlap between us and them when it came to supports