slic3r/Slic3r #4716: Adaptive Layer Height - Missing STL Features

maximusmountain, 2019-01-24:
Parts of layers missing when using adaptive layer heights, issue not present with fixed layer height

Steps to reproduce the behavior:
0.8mm Nozzle, 75% adaptive quality, min layer 0.2mm, max layer 0.55mm
When looking into slice preview there is parts of layers that are missing, this then causes some odd behaviour in the print. Shown in the .zip with a simple, and common, .stl file but problem persists with other geometries.

**Screenshots**
Screenshot added into .zip file.

**Desktop (please complete the following information):**
- Windows 10 64bit
 - Version 1.3.0

[Adaptive Slicing Bug 20190124.zip](https://github.com/slic3r/Slic3r/files/2793469/Adaptive.Slicing.Bug.20190124.zip)

supermerill, 2019-01-24:
Seams like a thin wall issue

maximusmountain, 2019-01-24:
> Seems like a thin wall issue

Yep... That's embarrassing, apologies.