slic3r/Slic3r #4723: Rectilinear option for metal printing

MikeMyersFMI, 2019-02-02:
I'm working on a 3D printing solution for use with a metal additive process. Due to the high thermal loads applied, I must deposit in long paths and stop at the end of each path to wait for the printed object to cool slightly before starting the next path. 

What I'd like to see is a rectilinear infill option (or another infill option) that gives me:
- 100% infill
- maximizes the length of a deposit given the geometry (minimizes the number of stops and starts)
- pauses the deposition between paths (beads) on the layer (i.e., does not print the entire layer in one path but breaks it down into segments).
- no perimeter 

See image below of what I have been able to achieve from the rectilinear infill. The short deposits on the short side of the L-shape will cause problems with my metal process. I need the beads to go lengthwise with the tall part of the L and also the short part of the L. Think of starting at the top of the L and depositing one continuous bead to the end of the short side of the L, pausing the deposition, positioning the head one bead-width to the side of the first deposition, and depositing another L-shape next the first, pausing the deposition again, and then repeating the process (in reverse order as far as deposit direction) for subsequent layers. 

[image]

seeul8er, 2019-02-08:
Increasing the number of perimeters (Num. perimeters > part width_height) might give you the desired result.
In the below image a increased them to 25 so basically there will be no infill as it is perimeters only.

[image]

supermerill, 2019-02-09:
And it's possible to have only one continuous unique extrusion loop, with the pr https://github.com/slic3r/Slic3r/pull/4695 (you may want to remove gap fill -> set gap fil speed to 0)