slic3r/Slic3r #4724: Missing "=" in config crashes slic3r

kisolre, 2019-02-04:
### Version
Slic3r-master.2019.01.25.2262.66b00c90.64bit-PR.zip

### Operating system type + version
Windows 8.1

### Behavior
When starting Slic3r.exe nothing shows and no error messages are displayed in console.
Checking with ProcessExplorer shows the process running for a second and then exits.
After some investigation I found that I had a broken ini file in Printers folder - truncated to 600b (which by itself does not cause an error) and missing "=" at the last line.
Running Slic3r-console.exe from command window gives the following error: 
`Caught C++ exception of type or derived from 'std::exception': <unspecified file>(2): '=' character not found in line at E:/3D Printing/Software/Slicer/slic3r/Slic3r/lib/Slic3r/Config.pm line 99.`

### Steps to reproduce: 
 Remove any "=" from a line where a variable is assigned value. Run slic3r - it exits without showing any errors.

### Expected  behavior
 Either ignore the faulty config and continue or ask the user for intervention.

dwillmore, 2019-02-04:
It tried to reproduce this by editing the slic3r.ini and it did not fail.  Then I tried it with a pritner .ini file (the currently selected printer) and it failed as described.  There may be a difference in the way the different files are parsed.  That might be a clue to get someone started.  

I had originally thought this might be an upstream error, but since the slic3r.ini can be parsed with these issues, there should be a way to solve this in slic3r.

kisolre, 2019-02-04:
I managed to break it also with slic3r.ini
[image]
Removing the "=" from last_output_path causes these errors
[image]
[image]

lordofhyphens, 2019-03-24:
@kisolre it'd be new settings up to where it failed.

PR #4765 (2019-04-12): Catch and rethrow parse errors, showing a log entry and a GUI error.
If a loaded config.ini has a parse error from boost::property_tree::read_ini, handle things a little more gracefully.

Fixes #4724
closed 2019-04-12 by lordofhyphens