slic3r/Slic3r #4726: "simplication" of contour polygons: remove collinear points from tesselation artifacts

platsch, 2019-02-05:
Consider a simple object like this:

[image]

The cylinder is composed of a number of rectangular surfaces, which is a very common pattern for many objects. The tesselated representation of this model looks like this:

[image]

All rectangular surfaces are now split into triangles. When slicing this model, the result looks like this (Layer 5):

[image]

Every additional triangle line (red marker in the tesselated image) in each rectangle causes slic3r to add a point which is perfectly collinear with its two neighbors, but entirely useless.
Most (but not all!) of the redundant points are removed during the gcode-export by the Douglas-Peucker algorithm (invoked via polygon.simplify [here](https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/GCode.cpp#L323), but all polygon operations during the entire slicing process have to respect the useless points.

Is there any specific reason to keep collinear polygon points? If not, I propose to remove them right after or during the `slice()` step in TriangleMeshSlicer.
I did an implementation of this, the result for the same object is as expected:

[image]

Some quick tests with different objects indicated a speedup of the full slicing process between 10% and 30%, but I did not extensively test this.  Worst case would be a model purely composed of triangles which do not assemble a larger, planar surface. In this case the `remove_collinear_points` would add some overhead.

I also noted a `remove_vertical_collinear_points` method in Polygon.cpp which is apparently used for rectilinear infill patterns, but I'm not sure whether a general implementation of remove_collinear_points could replace this.

If there is no obvious reason to not apply this simplification, I can quickly file a PR with my implementation.

lordofhyphens, 2019-02-05:
Probably fine, although there was related work for curve fitting.

platsch, 2019-02-05:
Can you point me somewhere? I'm not familiar with the curve fitting stuff

lordofhyphens, 2019-02-05:
https://github.com/slic3r/Slic3r/issues/23

platsch, 2019-02-05:
OK, sorry, I was actually aware of this thread. But I don't see how exactly it is related to this issue. I didn't look at the code in #23, is it because arc detection also has the problem of collinear points, preventing a correct detection? Or is there a solution for the problem I described in the arc-detection approach?

PR #4728 (2019-02-10): Simplify collinear points
Fixes #4726.

collinear lines are removed from the `slices` polygons of each layer (parallelized).

This could potentially also be implemented in the [make_loops](https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/TriangleMesh.cpp#L1135) method where the polygons from slicing are assembled, not sure if that would gain any performance.

Includes a simple CATCH testcase, but only for the remove_collinear_points part of Polygon.
commit faeb213c (2019-02-10): Remove collinear points from polygons right after slicing the stl into polygons.

Triangulation of planar surfaces results in additional edges (i.e. splitting a rectangle into two triangles).
Those edges end up as collinear points in the contour polygons and cause additional, redundant
computation. Most collinear points would be removed by Douglas-Peucker during gcode export.

Fixes #4726.
closed 2019-02-10 by lordofhyphens