slic3r/Slic3r #4730: changing the title of filament settings tab

farhadcomp, 2019-02-07:
Hi,
I was trying to edit the title of filament settings tab in slic3r with editing the sub title( 'Filament Settings') in PresentEditor.pm but noting happened
[image]
.
how can I edit the title of filament settings tab?

lordofhyphens, 2019-02-08:
Probably in MainFrame.pm, although I have no idea why you want to do this.

farhadcomp, 2019-02-08:
> Probably in MainFrame.pm, although I have no idea why you want to do this.

Thanks for your reply @lordofhyphens.
I am trying to specialize filament settings for special material and I want to show this material instead of filament.
I couldn't find how can I edit filament settings title in MainFrame.pm. It is simple to edit title of Plater Tab    in MainFrame.pm just by editing the "Plater" in sub init_tabpanel:

>     $panel->AddPage($self->{plater} = Slic3r::GUI::Plater->new($panel), "Plater");

but it is not the same for Filament Settings title.

lordofhyphens, 2019-02-08:
That is already going to change in the C++ GUI to material.

I haven't looked at that area of the code in months.

lordofhyphens, 2019-03-24:
@farhadcomp I went digging, and the tabs have their names set in a different place than the dialog boxes. https://github.com/slic3r/Slic3r/blob/21eb603cc16946b14e77d3c10cbee2f1163503c6/lib/Slic3r/GUI/Plater.pm#L901

closed 2019-03-24 by lordofhyphens