slic3r/Slic3r #4731: Master branch isn't compiled after wxwidgets

Splarv, 2019-02-08:
macOs Mojave 10.14.2
Soft installed by instructions (homebrew variant):
https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-OS-X
$ git status -v
On branch master
Your branch is up to date with 'origin/master'.
$ git remote -v
origin	https://github.com/slic3r/Slic3r.git (fetch)
origin	https://github.com/slic3r/Slic3r.git (push)
$ git branch -v

* master [66b00c9](https://github.com/slic3r/Slic3r/commit/66b00c90c3121cf0e67ea7a345b42f27098fefe4) Fix typo.

[build.log.txt](https://github.com/slic3r/Slic3r/files/2845716/build.log.txt)

lordofhyphens, 2019-03-24:
I'm not going to try to troubleshoot building wxmac, it's enough of a headache figuring out what is going on with my system much less anyone else's. Use the homebrew build (which is what the distribution uses).

https://formulae.brew.sh/formula/wxmac

closed 2019-03-24 by lordofhyphens

Splarv, 2019-03-25:
> I'm not going to try to troubleshoot building wxmac, it's enough of a headache figuring out what is going on with my system much less anyone else's. Use the homebrew build (which is what the distribution uses).
> 
> https://formulae.brew.sh/formula/wxmac

I did exactly what was written in the developer documentation for slic3r. To fix the bug need fix the build process or fix the build manual. Not just close the bug. :)