slic3r/Slic3r #4732: width settings % in function of nozzle diameter instead of layer height

supermerill, 2019-02-08:
I don't know how a width in a % of a height can be useful.

At least, with the nozzle diameter, it can adapt dynamically to the nozzle.

I've done that for my fork and it's a really little change.

Opinions?

VanessaE, 2019-02-08:
I've never understood it either, but "width over height" became an extremely common term early on, and is integral to older slicers like Skeinforge.  Maybe in Slic3er, it's used in the setting of line widths when they're set to "auto" and/or "default" (in Print settings -> Advanced)?

supermerill, 2019-02-08:
when you put 0 in the width, it compute the width from the nozzle diameter.

lordofhyphens, 2019-02-09:
#3658 for @alranel comment on it.

lordofhyphens, 2019-02-09:
It is a change that can and will break people's configs in unexpected ways (as it is a change to the semantics of an option). I wouldn't want to just change it without providing some mitigation for previous configs or making it an additional keyword flag. 

https://reprap.org/wiki/Width_over_height
https://pastebin.com/2LK0SGAy

On the other hand, apparently @daid has feelings on this ratio.
https://community.ultimaker.com/topic/1798-width-over-height-setting/

https://github.com/daid/LegacyCura/issues/306