slic3r/Slic3r #4733: Slic3r experiment - Create gaps within the capital to create a v shape

JamesVerner, 2019-02-08:
Hey guys, this should be my last question regarding experimental possibility with Slic3r. However, I am wondering would anyone know how the following could be achieved. I know within the code base:

https://github.com/slic3r/Slic3r/blob/66b00c90c3121cf0e67ea7a345b42f27098fefe4/lib/Slic3r/Print/SupportMaterial.pm#L953

It is possible to increase the capital, which I imagine is the name given to the area that is above the pillar and is in contact with the 3D object. That being said, is it possible to also remove some of the material within this area as well? This picture below shows what I mean, a Y shape with the support going up and then splitting into two.

[image]

Any guidence would be great, thanks!

supermerill, 2019-02-09:
why V shape and not a diamond one?

JamesVerner, 2019-02-09:
More of a what if to be honest. To see the possibility of moving this to
some form of branching technique.

lordofhyphens, 2019-02-09:
You would have to modify the logic for how the projection down from the contact area is done, layer-by-layer drawing the polygon area.

JamesVerner, 2019-02-09:
Hey Joseph, would you have an idea where that might be? I know the contact
is mentioned on a number of occasions throughout the supportmaterial.pm. If
we are talking about downwards, the only thing that strikes me (since it is
mentioned in the comment) would be this:

https://github.com/slic3r/Slic3r/blob/66b00c90c3121cf0e67ea7a345b42f27098fefe4/lib/Slic3r/Print/SupportMaterial.pm#L64

That being said, so far anything I have done with changing the shape of the
existing pillars has been done in the pillar method itself.

https://github.com/slic3r/Slic3r/blob/66b00c90c3121cf0e67ea7a345b42f27098fefe4/lib/Slic3r/Print/SupportMaterial.pm#L915

Thanks.