slic3r/Slic3r #4734: Can't install Slic3r on Raspberry Pi 3 Model B+

NovaViper, 2019-02-09:
**Describe the bug**
After following this [wiki](https://plugins.octoprint.org/plugins/slic3r/), I can't seem to get slic3r to build at all. I keep getting an error like this:
`arm-linux-gnueabihf-gcc -I/usr/lib/arm-linux-gnueabihf/perl/5.24/CORE -fPIC -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/libslic3r/LayerRegion.o src/libslic3r/LayerRegion.cpp
In file included from src/libslic3r/Flow.hpp:5:0,
                 from src/libslic3r/Layer.hpp:5,
                 from src/libslic3r/LayerRegion.cpp:1:
src/libslic3r/Config.hpp: In member function ‘virtual bool Slic3r::ConfigOptionFloat::deserialize(std::__cxx11::string)’:
src/libslic3r/Config.hpp:68:29: error: cannot convert ‘std::basic_istream<char>::__istream_type {aka std::basic_istream<char>}’ to ‘bool’ in return
         return iss >> this->value;
                             ^~~~~
src/libslic3r/Config.hpp: In member function ‘virtual bool Slic3r::ConfigOptionInt::deserialize(std::__cxx11::string)’:
src/libslic3r/Config.hpp:127:29: error: cannot convert ‘std::basic_istream<char>::__istream_type {aka std::basic_istream<char>}’ to ‘bool’ in return
         return iss >> this->value;
                             ^~~~~
src/libslic3r/Config.hpp: In member function ‘virtual bool Slic3r::ConfigOptionPercent::deserialize(std::__cxx11::string)’:
src/libslic3r/Config.hpp:252:29: error: cannot convert ‘std::basic_istream<char>::__istream_type {aka std::basic_istream<char>}’ to ‘bool’ in return
         return iss >> this->value;
                             ^~~~~
src/libslic3r/Config.hpp: In member function ‘virtual bool Slic3r::ConfigOptionFloatOrPercent::deserialize(std::__cxx11::string)’:
src/libslic3r/Config.hpp:282:29: error: cannot convert ‘std::basic_istream<char>::__istream_type {aka std::basic_istream<char>}’ to ‘bool’ in return
         return iss >> this->value;
                             ^~~~~
error building src/libslic3r/LayerRegion.o from 'src/libslic3r/LayerRegion.cpp' at /usr/share/perl/5.24/ExtUtils/CBuilder/Base.pm line 174.
FAIL
! Installing ./xs failed. See /root/.cpanm/work/1549718319.25671/build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting
`

**To Reproduce**
Steps to reproduce the behavior:
1. Follow instructs listed on the Octoprint plugin website
2. After running the script, notice the error
4. See error

**3D Model and Slic3r Configuration Export**
N/A

**Expected behavior**
Should have finished building without any errors

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Raspberry Pi 3 terminal/Linus
 - Version 1.2.9/1.3.0

**Additional context**
I eventually made the install script work after using force, but for some reason, using it for Build.PL doesn't seem to work at all to help the problem

lordofhyphens, 2019-02-14:
which compiler?

lordofhyphens, 2019-02-14:
Also, can you try to build with 1.3.1-dev instead?

NovaViper, 2019-02-17:
I'm using whatever compiler that came with the OctoPi image. I haven't tried 1.3.1-dev however, I'm going to try that now

NovaViper, 2019-02-17:
Still getting this error:

`
Configuring threads-shared-1.59 ... OK
Building and testing threads-shared-1.59 ... OK
Successfully installed threads-shared-1.59 (upgraded from 1.51)
1 distribution installed
cpanm (App::cpanminus) 1.7044 on perl 5.024001 built for arm-linux-gnueabihf-thread-multi-64int
Work directory is /root/.cpanm/work/1550381553.16757
You have make /usr/bin/make
You have /usr/bin/wget
You have /bin/tar: tar (GNU tar) 1.29
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by John Gilmore and Jay Fenlason.
You have /usr/bin/unzip
--> Working on ./xs
Entering /home/pi/Slic3r/xs
Checking if you have ExtUtils::Install 1.46 ... Yes (2.06)
Checking if you have Module::Build 0.38 ... Yes (0.4220)
Running Build.PL
Configuring /home/pi/Slic3r/xs ... Slic3r requires the Boost libraries. Please make sure they are installed.

If they are installed, this script should be able to locate them in several
standard locations. If this is not the case, you might want to supply their
path through the BOOST_INCLUDEPATH and BOOST_LIBRARYPATH environment variables:

    BOOST_INCLUDEPATH=/usr/local/include BOOST_LIBRARYPATH=/usr/lib perl Build.PL

If you just compiled Boost in its source directory without installing it in the
system you can just provide the BOOST_DIR variable pointing to that directory.

N/A
! Configure failed for ./xs. See /root/.cpanm/work/1550381553.16757/build.log for details.
The XS/C++ code failed to compile, aborting
If everything was installed properly,you should be able to run Slic3r with the command ./slic3r.pl

`

lordofhyphens, 2019-02-18:
@NovaViper So how are you trying to build now? From the wiki instructions?

NovaViper, 2019-02-18:
Yeah, from the wiki instructions

lordofhyphens, 2019-02-19:
Configure says it can't find boost. You can build boost separately on the system and use that as an environment variable for configure

lordofhyphens, 2019-02-19:
I don't have a raspi3 on hand to test with yet so I can't do much else until then

NovaViper, 2019-02-19:
How do I setup Boost?

lordofhyphens, 2019-02-19:
Download the source and compile it.

NovaViper, 2019-02-19:
Where do I find the library at? And how do I put it on the raspberry pi?

lordofhyphens, 2019-02-19:
From Google:
https://www.boost.org/doc/libs/1_69_0/more/getting_started/unix-variants.html

Normally this should installable from your package manager (the associated -dev packages for boost)

NovaViper, 2019-02-19:
Thanks for the info. When I get back to my house this weekend i'll go and see if I can get the program installed :D

NovaViper, 2019-02-23:
Ok, so I figured out how to get it to properly compile after looking at [this wiki](https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi) and got it all working!

closed 2019-02-23 by NovaViper

lordofhyphens, 2019-02-23:
If you'd like to put your own notes up on ours (or at least a link to the
octoprint wiki), it'd be appreciated.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

NovaViper, 2019-02-23:
I basically followed the instructions on the OctoPrint Slic3r plugin's Github wiki (https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi) and used the settings for the last known working version for Slic3r PE. Although.. I'm having issues slicing objects as it says something was wrong with my gcode