slic3r/Slic3r #4736: problem on loading slic3r.exe

farhadcomp, 2019-02-11:
hi everyone,
when I made a zip file as it was said in [https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-Windows](url) I got a message: "can not find freeglut.dll" but after that it created zip file without any errors:
[image]

but my problem is when I doubled click on slic3r.exe from unpacked zip folder,  nothing happened and slic3r don't open. here is unpacked zip folder:
[image]

so what is the problem? does it happen just because of missing freeglut.dll?

lordofhyphens, 2019-03-02:
You're missing freeglut.dll, which is available online (it's an open-source library and not strictly part of slic3r).

lordofhyphens, 2019-03-02:
Note that CLI slicing will still work, freeglut is just a requirement for the GUI

closed 2019-03-02 by lordofhyphens