slic3r/Slic3r #4738: Configure Failed for ./xs 1.2.9

TolleyB-J, 2019-02-13:
**Describe the bug**
There is a compliling error with the XS library

**To Reproduce**
Steps to reproduce the behaviour:
1. Install OctoPi
2. Run this script typing in 1.2.9 as the version when it asks
```
 #!/bin/sh
  
 echo "Shell script created by Javier Martínez Arrieta for Slic3r installation\n"
 #Ask the user for a version
 echo "Please indicate which version you desire to be installed (e.g. 1.1.7)"
 read version
 echo "The installation of Slic3r takes a long time. PLease be patient"
 cd $HOME
 echo "Installing required libraries and dependencies..."
 sudo apt-get install git libboost-system-dev libboost-thread-dev git-core build-essential libgtk2.0-dev libwxgtk2.8-dev libwx-perl libmodule-build-perl libnet-dbus-perl cpanminus libextutils-cbuilder-perl gcc-4.7 g++-4.7 libwx-perl libperl-dev
 sudo cpanm AAR/Boost-Geometry-Utils-0.06.tar.gz Math::Clipper Math::ConvexHull Math::ConvexHull::MonotoneChain Math::Geometry::Voronoi Math::PlanePath Moo IO::Scalar Class::XSAccessor Growl::GNTP XML::SAX::ExpatXS PAR::Packer
 echo "Cloning Slic3r repository..."
 git clone https://github.com/alexrj/Slic3r.git
 cd Slic3r
 git checkout $version
 echo "Building and testing Scli3r..."
 sudo perl Build.PL
 echo "If everything was installed properly,you should be able to run Slic3r with the command ./slic3r.pl"
```

**Log Files**
Console:
```
Shell script created by Javier Martínez Arrieta for Slic3r installation\n
Please indicate which version you desire to be installed (e.g. 1.1.7)
2.1.9
The installation of Slic3r takes a long time. PLease be patient
Installing required libraries and dependencies...
Reading package lists... Done
Building dependency tree
Reading state information... Done
build-essential is already the newest version (12.3).
cpanminus is already the newest version (1.7042-2).
libmodule-build-perl is already the newest version (0.422000-1).
libmodule-build-perl set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgtk2.0-dev : Depends: libglib2.0-dev (>= 2.27.3) but it is not going to be installed
                 Depends: libgdk-pixbuf2.0-dev (>= 2.21.0) but it is not going to be installed
                 Depends: libpango1.0-dev (>= 1.20) but it is not going to be installed
                 Depends: libatk1.0-dev (>= 1.29.2) but it is not going to be installed
                 Depends: libcairo2-dev (>= 1.6.4-6.1) but it is not going to be installed
                 Recommends: debhelper but it is not going to be installed
 libnet-dbus-perl : Depends: libxml-twig-perl but it is not going to be installed
 libperl-dev : Depends: perl:any (= 5.24.1-3+deb9u5)
               Depends: libperl5.24 (= 5.24.1-3+deb9u5) but 5.24.1-3+deb9u3 is to be installed
 perl-modules-5.24 : Breaks: libextutils-cbuilder-perl (< 0.280225)
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
--> Working on AAR/Boost-Geometry-Utils-0.06.tar.gz
Fetching http://www.cpan.org/authors/id/A/AA/AAR/Boost-Geometry-Utils-0.06.tar.gz ... OK
Configuring Boost-Geometry-Utils-0.06 ... OK
Building and testing Boost-Geometry-Utils-0.06 ... OK
Successfully installed Boost-Geometry-Utils-0.06
Math::Clipper is up to date. (1.27)
Math::ConvexHull is up to date. (1.04)
Math::ConvexHull::MonotoneChain is up to date. (0.01)
Math::Geometry::Voronoi is up to date. (1.3)
[... 107 lines cut ...]
Work directory is /root/.cpanm/work/1550082186.9177
You have make /usr/bin/make
You have /usr/bin/wget
You have /bin/tar: tar (GNU tar) 1.29
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by John Gilmore and Jay Fenlason.
You have /usr/bin/unzip
--> Working on ./xs
Entering /home/pi/Slic3r/xs
Checking if you have ExtUtils::Install 1.46 ... Yes (2.06)
Checking if you have Module::Build 0.38 ... Yes (0.4220)
Configuring /home/pi/Slic3r/xs
Running Build.PL
-> N/A
-> FAIL Configure failed for ./xs. See /root/.cpanm/work/1550082186.9177/build.log for details.
```

**Desktop :**
 - OS: Rasbian
 - Version: 1.2.9
 - Device; Raspberry Pi 3

lordofhyphens, 2019-02-14:
Looks like the script you are using is ancient and won't help much. You should do what the config error is telling you (and install all of the headers instead of just one).

Libslic3r also needs gcc5 or newer (c++11 support)

Take a look at the Linux instructions on the wiki. If you have to do something different, then add a raspi section and write down what you did.

lordofhyphens, 2019-02-14:
We will not support someone's script to compile Slic3r (you should ask the author directly).

lordofhyphens, 2019-02-14:
Will reopen if you can show an actual build error, not Build.PL telling you to install prerequisites.

As I said earlier, see wiki: https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux

If you found that you had to install a couple more packages, add them to the wiki

closed 2019-02-14 by lordofhyphens