slic3r/Slic3r #4739: Hi, after installation of  Win10-update on patchday Feb 13th slic3r opens and works extremely slow,

cypcyp51, 2019-02-14:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**3D Model and Slic3r Configuration Export**
Please upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. MacOS, Linux, Windows]
 - Version [e.g. 1.3.0]
     - _Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._
     - _For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
     - _Use `About->About Slic3r` for release versions._

**Additional context**
Add any other context about the problem here.

cypcyp51, 2019-02-14:
Sorry, published the question before finishing. After removal of the update KB4481031 slic3r 1.3.0 worked fine again until the update installs again.

supermerill, 2019-02-14:
Last time it was the graphic drivers. See if windows doesn't update/downgrade your driver with the update. Or just install a good one.

cypcyp51, 2019-02-14:
Already looked at the graphics driver, seem to be fine, it's the intel HD Graphics 4000 from Dec 21. 2015, so nothing has changed

cypcyp51, 2019-02-14:
Meanwhile i checked intel for new graph drivers and installed the driver dated Aug 16th, 2018. Since i have to work with slic3r today, i have blocked the re-installation of updates. As soon as i have time left i will retry with the new updates for win10 again. 
On my notebook with win10 1803 the updates from Feb 13th seem to have no effect on the program.

lordofhyphens, 2019-02-14:
Odd, KB4481031 should have zero effect (it was a .NET update, which Slic3r doesn't use).

The other KB isn't the full string so I don't know what it is.

cypcyp51, 2019-02-14:
Sorry, had forgotten to enter 2 digits, KB4483452, was also .NET update. Probably i fooled myself with the de-installed updates.
Meanwhile i have installed all updates again for test purposes - und all works fine again - so far.
Probably the outdated Intel graphs driver didn't want to work with some other update features.
Thanks a lot for the hints.

closed 2019-02-14 by supermerill