slic3r/Slic3r #4741: Rotation buttons are not working

thiagopeixoto16, 2019-02-14:
These four rotation buttons at GUI are not working:
[image]

Only Z direction is working fine. The other (x and y) you click and nothing happens.

 - OS: Windows 10
Version 1.3.1 dev 2019.02.10 64 bits from automated builds

foreachthing, 2019-02-15:
Funny ... in Slic3r-master.2019.02.10.2284.05a2f902.64bit-PR I don't see any rotation buttons, if preference for "Rotation controls in toolbar" is set to "Z only".
But with the option "XYZ", I get all the icons and those functions are working.

I haven't changed anything ... so, I don't know who has or what happened.... anyone?

thiagopeixoto16, 2019-02-15:
Thanks for the inputs. In this case let me show more information in order to reproduce the problem:
This is my Preferences:
[image]
I've tested with several files, but you can have this one as example:
[iPhone5Case6.zip](https://github.com/slic3r/Slic3r/files/2869166/iPhone5Case6.zip)

I'll be glad to provide any information required.

foreachthing, 2019-02-18:
oh, try to set a Color Scheme to non empty - this _might_ help.

thiagopeixoto16, 2019-02-18:
Hi @foreachthing, did you tried to remove and the error occurred or just give a blind hunch?
Anyway I tried here and the problem persists.

foreachthing, 2019-02-19:
Blind hunch, definitely! Empty settings are ... weird.
Well, I tried it again, at home on Win10, and only the Z-rotation works.
The rotation commands from the context menu work as expected....

foreachthing, 2019-02-19:
Found an old version (Slic3r-master.2018.06.28.1720.12aa06e.64bit) and that one (still) works with Win10.

thiagopeixoto16, 2019-03-04:
@lordofhyphens I think this can be tagged as a verified bug.