slic3r/Slic3r #4742: 'Tool Change Gcode' in 'Custom G-code' section does not appear in final gcode file

joonadec, 2019-02-19:
**Describe the bug**
Slic3r does not include any of the Gcode defined in the 'Tool Change' text box, only the 'Before Layer Change' and 'After Layer Change' boxes.

**To Reproduce**
I've generated a two-colour dual-material print. The before/after layer Gcode bits get added into the file, but whatever I put into the 'Tool Change' box doesn't get included.

**3D Model and Slic3r Configuration Export**
Included 'Traffic Cone' STL and gcode

[Traffic_Cone_Rings_White_0.15mm_PLA_MK3S.zip](https://github.com/slic3r/Slic3r/files/2882349/Traffic_Cone_Rings_White_0.15mm_PLA_MK3S.zip)

**Expected behavior**
Tool change gcode should appear after the 'Before Layer Change' and before the 'After Layer Change' sections

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 7
 - Version 1.41.3

**Additional context**
Could not find any checkboxes to enable/disable tool change Gcode (thinking perhaps it is not included by default)

lordofhyphens, 2019-02-20:
Reproduce with a build from dl.slic3r.org/dev

Your version number indicates you are using the Prusa3d fork. 

If you can reproduce with a build from dl.slic3r.org/dev then I will reopen. Otherwise you are instructed to go to prusa3d/Slic3r (which is what the text you removed to create this bug report says)

closed 2019-02-20 by lordofhyphens

lordofhyphens, 2019-02-20:
Just to confirm: I cannot reproduce with current development version of Slic3r, so this definitely be should go to prusa3d/Slic3r.

@joonadec In the future, it is asked that you provide a config export (Ctrl-E). I tried to reproduce your setup best as I could, but it would be more helpful (to the Prusa3d devs) that you use the AMF or 3MF export to export the combined file (as I include). 

My reproduction setup: [4742.zip](https://github.com/slic3r/Slic3r/files/2885514/4742.zip)

@bubnikv just a FYI for this one.