slic3r/Slic3r #4744: Overhang Detection - How does it actually work

JamesVerner, 2019-02-24:
I've spent a fair amount of time trying to work out how to make different shapes within Slic3r, unfortunately still working on branching but I've realised my knowledge of where the supports are generated is non-existent. Looking at the code base:

https://github.com/slic3r/Slic3r/blob/21eb603cc16946b14e77d3c10cbee2f1163503c6/lib/Slic3r/Print/SupportMaterial.pm#L150

I know providing a value to the overhang threshold tells the program to detect the places to provide support. Within that code it can be found two options for detecting overhangs, I know options are mentioned in #2340, the one I am referring to is this:

https://github.com/slic3r/Slic3r/blob/21eb603cc16946b14e77d3c10cbee2f1163503c6/lib/Slic3r/Print/SupportMaterial.pm#L164

I know there is a fair bit to go, but that is all I know so far. The question I am asking, is how does the program determine which area is the best to create support and where is it? How does it also detect overhangs? (I know 45 degrees is the critical angle calibation but I know nothing else) Thanks guys for all the previous answers as well.

supermerill, 2019-02-25:
imo, i think it print support for diff(next layer, current layer). 

offset = lower_layer.height / tan(overhang_setting)

support area = diff( next_layer, offset(current_layer, offset) )

JamesVerner, 2019-02-25:
Hey Merill, thanks for replying. However I am just wondering, how does it
detect the area where it needs to make the support? Like in MeshMaker it
shows the area where they would build the support, how does Slic3r carry
out the same question?

supermerill, 2019-02-25:
that's what i said.

you have slices.

you look two slices, and make a diff.

the result is the unsupported area, where you need to have support.

JamesVerner, 2019-02-25:
Thanks again Merill. I'm sorry but I still don't understand, do you mean that you would you go up layer by layer storing each slice and comparing the difference?

lordofhyphens, 2019-02-25:
At the time we are calculating support, we already have all of the sliced model layers.

What the usual algorithm does is look at how much island A on layer 1 is bigger than the same island on layer 2. If it is above a certain size then we mark it.

JamesVerner, 2019-02-25:
Hey Joseph, thanks for responding. Just quick follow up questions:
- Is the algorithm you are referring to the one located here:
https://github.com/slic3r/Slic3r/blob/21eb603cc16946b14e77d3c10cbee2f1163503c6/lib/Slic3r/Print/SupportMaterial.pm#L164

- Likewise, if we are comparing each island from layer to layer. When do you exactly is the support introduced? Is that where the threshold comes into play? If that is the case, where would that be located sorry? 

- Also sorry about this last question, where exactly are the sliced layers stored at in this file where the algorithm is located at?

JamesVerner, 2019-02-26:
I know you mentioned Merill regarding that offset,
offset = lower_layer.height / tan(overhang_setting). However while the first part makes sense, not too sure what you mean by the tan(overhang_setting) unless that is the threshold.

lordofhyphens, 2019-03-24:
@JamesVerner it seems to work fine in most cases and 45 deg is going to require fewer layers in Z to achieve the same horizontal displacement than 50 deg, which is why we'd default to it.

https://github.com/slic3r/Slic3r/blob/21eb603cc16946b14e77d3c10cbee2f1163503c6/lib/Slic3r/Print/SupportMaterial.pm#L398 returns the layers generated for the support layers (but not interface, that's a separate method).

lordofhyphens, 2019-03-24:
Support isn't calculated until the model is completely sliced (this is why it's more difficult than it would be otherwise to turn them on/off with modifier meshes; such an implementation needs to translate the 3D space mesh into a series of slices and then overlay those with the model slices).