slic3r/Slic3r #4745: Slic3r takes forever to build

marcel2034, 2019-02-27:
**Bug description**
Slic3r is very fast to build with the C++ method 
```
mkdir build  
cd build   
cmake ../src/   
cmake --build . -- -j4   
```
 but ./slic3r returns unknown exception 
as in issue : [4719](https://github.com/slic3r/Slic3r/issues/4719). 

When building with perl, slic3r runs, but it takes a very long time (15 min) to build, even with small modifications. (If I want to play with the code, it won't be viable)

Is there something I'm missing ? 
Can I boost perl building ? 
Should I be able to run slic3r with just the C++ side ? 

**System**
- OS: Ubuntu
- Version 18.04
- 1.3.0-1302-g05a2f902

supermerill, 2019-02-27:
it's because, with perl, you rebuild everything each time.

After a full build, if you only do a small modification in a source file, you can:

cd xs
./build install

i think you also need to export something for this to work, but i don't remember what.

lordofhyphens, 2019-02-27:
@supermerill you need to manually set local::lib path (see local::lib documentation on cpan)

marcel2034, 2019-02-28:
@supermerill  @lordofhyphens Thanks for replying ! 
I'm not sure I understand where to set the local::lib path. 

I tried to add the local::lib command in .bashrc : 
```eval "$(perl -I$HOME/perl5/lib/perl5 -Mlocal::lib)"```
but it doesn't seem to work. 

I can `./Build install` in xs folder, but the code changes don't appear in the execution of slic3r.

lordofhyphens, 2019-02-28:
eval "$(perl -I$HOME/perl5/lib/perl5 -Mlocal::lib=/path/to/slic3r/local-lib)"

ben2034, 2019-02-28:
It worked :) 
Thank you very much !

marcel2034, 2019-02-28:
It worked :) 
Thank you very much !

closed 2019-02-28 by marcel2034