slic3r/Slic3r #4747: optimize extruder movement to exclude unnecessary moves across the layer (extruder on the end of one element starts with nearest element)

woodyv, 2019-02-28:
### Do not request features for the Prusa3D fork of Slic3r here without confirming it is applicable to the mainline Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

supermerill, 2019-02-28:
https://en.wikipedia.org/wiki/Travelling_salesman_problem

woodyv, 2019-03-01:
yes I know, but wave algoritm (as 4example in dps placement and routing) is good enough. in a simplest way it is posible to sort g commands in nearest end/begin point before output to the file.