slic3r/Slic3r #4749: Printing support before Skirt

Maxwellfire, 2019-03-02:
When support material generation is enabled, and a skirt is enabled, the skirt is printed after the support material. This means that we lose the priming effect of the skirt and the support material often prints badly. I would understand this problem if the skirt was a larger layer height than the first layer of the model/support, but this occurs even when that's not the case.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable support
2. Enable a skirt 
3. Export Gcode and see that the support is printed before the skirt

Config:
https://pastebin.com/CNZJYLc0

**Expected behavior**
Support would print after skirt so nozzle is primed

**Screenshots**
"First layer" in toolpath preview (no skirt shown):
[image]

"Second Layer"
[image]

 - OS: Windows
 - Version 1.3.0-1302-g05a2f902

lordofhyphens, 2019-03-24:
This is probably because the support is printing thinner than the rest of the material on the layer.

lordofhyphens, 2019-03-24:
@Maxwellfire Where did you get your build from? I can't find that hash anywhere and the filename looks really really old.

lordofhyphens, 2019-03-24:
[3D Model and configuration](https://github.com/slic3r/Slic3r/files/2999970/4749.zip)
[output gcode](https://github.com/slic3r/Slic3r/files/2999969/problem-case.txt)

Looks like that's exactly what is happening, for one reason or another, the skirt is at z=0.204mm and the support is at z=0.2mm.

In the future, please include a 3D model you see the problem with.

Maxwellfire, 2019-03-24:
Sorry about not including a model. I see the issue with 100% of models that have support enabled on them.

This might be because of the Z fulls steps/mm that I have set for my leadscrews (83.33)? Slic3r is rounding the layer height, but not rounding the support material layer height, and so the supports are printed on a slightly lower layer than the skirt and rest of the model? Based on the numbers you reported, this feels likely to me.

For the version that I'm using, I realize now that the hash included some local changes that hadn't been committed. I can't imagine how they would cause this problem (They are a small change in CoolingBuffer), but if it would help I can confirm that the bug still exists on the lastest master commit (05a2f9028). That is the hash of the commit I built (plus that 1 local change).

Maxwellfire, 2019-03-24:
Confirmed: if I disable the Z full steps/mm, the issue goes away.

lordofhyphens, 2019-03-24:
Interesting, that would indicate that support isn't getting the z full steps applied. I don't think that will be too hard to write a test for.

I did reproduce it on https://github.com/slic3r/Slic3r/commit/05a2f9028a35c56e039cf7a531d63b7ff7e97645 (hence my gcode link).

lordofhyphens, 2019-03-24:
The hash threw me for a loop because @RuiCarneiro put the exact same version hash plus that old build number on their issue as well (#4764). I was wondering where two people had found the same build that didn't match any commits in Slic3r.

Maxwellfire, 2019-03-24:
Oh, I think I have an idea what's going on with the hash. 

If you call `git describe --tags`, it outputs `1.3.0-1302-g05a2f902` for me. 1.3.0-1302-g**05a2f902** includes the correct hash after the g. I don't know why that extra g is included, but it is.

Maxwellfire, 2019-03-24:
From the git documentation:

> With something like git.git current tree, I get:
> 
>`[torvalds@g5 git]$ git describe parent`
> `v1.0.4-14-g2414721`
> 
> i.e. the current head of my "parent" branch is based on v1.0.4, but since it has a few commits on top of that, describe has added the number of additional commits ("14") and an abbreviated object name for the commit itself ("2414721") at the end.
> 
> The number of additional commits is the number of commits which would be displayed by "git log v1.0.4..parent". The hash suffix is "-g" + 7-char abbreviation for the tip commit of parent (which was 2414721b194453f058079d897d13c4e377f92dc6). **The "g" prefix stands for "git"** and is used to allow describing the version of a software depending on the SCM the software is managed with. This is useful in an environment where people may use different SCMs.
>

lordofhyphens, 2019-03-25:
@maxwellfire one of the reasons I think it is better to use `git rev-parse --short HEAD` (which is what the download uses) is that it should be consistent.

Maxwellfire, 2019-03-25:
Could you possibly change the issue template to recommend that? I got the command to run from there, and RuiCarneiro probably did as well.

lordofhyphens, 2019-03-25:
I thought that I had. I will change the template